mcu-tools/mcuboot #959: Upgrade issues when alignments of partitions differ

utzig, 2021-03-09:
While trying to upgrade using the primary partition in internal flash and secondary+scratch in an external flash, the upgrade fails. The flashes are configured with different alignment sizes (4 for the internal, 8 for the external).

After a bit of debugging the issue seems to be because 8 is used as the "global" write_sz, while the status area size for the primary partition is still calculated based on the alignment of the flash where it is stored, resulting in an overrun.

d3zd3z, 2021-03-10:
Is a reasonable fix to make sure we always use the largest alignment of the partitions?

utzig, 2021-03-10:
> Is a reasonable fix to make sure we always use the largest alignment of the partitions?

Yes, as long as this alignment is used everywhere.

closed 2021-06-03 by github-actions[bot]