mcu-tools/mcuboot #969: How to encrypt binary body, not only sign

Cherish-forever, 2021-03-25:
I am running 
`$ ./bootloader/mcuboot/scripts/imgtool.py sign --key bootloader/mcuboot/root-rsa-2048.pem --header-size 0x200 -S 0x20000 --align 8 --version 1.3 -E build_space/hello_world/black_f407zg_pro/zephyr/zephyr.hex signed-zephyr-1.3.hex`

and I found that binary body does not encrypt, only add a header and tail, but body in middle was not encrypt.
what should I encrypt the binary body(code and data).

utzig, 2021-03-25:
You must pass a public key to `-E`, no the firmware file.

Cherish-forever, 2021-03-25:
@utzig add  -E bootloader/mcuboot/root-rsa-2048.pem ?

Cherish-forever, 2021-03-25:
@utzig 
it works
Thank you very much,  but I don`t understand why encrypt do not use --key bootloader/mcuboot/root-rsa-2048.pem file?

Cherish-forever, 2021-03-30:
@utzig 
Thank you very much, I have encrypt firmware by adding -E parameter.
but, I found after encrypted, images sha256 verify failed.
sign key is different with  body encrypt key, but the same kind of encrypt type.

closed 2021-06-14 by github-actions[bot]