mcu-tools/mcuboot #970: Undefined boot_write_trailer fns

mjahamel, 2021-03-29:
I am trying to build the current MCUboot for zephyr/nrf52833 following https://github.com/mcu-tools/mcuboot/blob/master/docs/readme-zephyr.md, and I'm running into some sort of linker issue: the board make step succeeds, but ninja fails with:

/home/michael/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: app/libapp.a(bootutil_misc.c.obj): in function `boot_write_copy_done':
/home/michael/mcuboot/boot/bootutil/src/bootutil_misc.c:306: undefined reference to `boot_write_trailer_flag'
/home/michael/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: app/libapp.a(bootutil_misc.c.obj): in function `boot_write_swap_size':
/home/michael/mcuboot/boot/bootutil/src/bootutil_misc.c:318: undefined reference to `boot_write_trailer'
collect2: error: ld returned 1 exit status

This is reproducible from a fresh git clone, so I don't believe its anything I have done. I would welcome any advice.

nvlsianpu, 2021-03-29:
@mjahamel What zephyr version you are using?

mjahamel, 2021-03-29:
The current Zephyr: 2.5.99

nvlsianpu, 2021-03-30:
It does work for me. I just build nrf52833dk_nrf52833. I have zephyr checkout'ed at https://github.com/zephyrproject-rtos/zephyr/commit/ae4a28ca08ee69dba72f0d93cc2ca311d41bf885. MCUboot checkout'ed at master HEAD of this repo (be aware that zephyr is using mirror fork of this repo, I'm synchronizing them more-less bi-weekly).

Despite readame-zephyr.md describe building process well, you can build MCUboot using `west` comadset as well:
assumed `pwd` gives the `zephyr-project` directory, which contains the zephyr repository in `zephyr` directory:
``` console
anpu@ubuntu:~/codedev/zephyrproject$ source zephyr/zephyr-env.sh 
anpu@ubuntu:~/codedev/zephyrproject$ west build -d build/mcuboot_nrf52833 -b nrf52833dk_nrf52833 bootloader/mcuboot/boot/zephyr/
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/anpu/codedev/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/anpu/codedev/zephyrproject/bootloader/mcuboot/boot/zephyr
-- Zephyr version: 2.5.99 (/home/anpu/codedev/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components:  Interpreter 
-- Found west (found suitable version "0.9.0", minimum required is "0.7.1")
-- Board: nrf52833dk_nrf52833
-- Cache files will be written to: /home/anpu/.cache/zephyr
-- Using toolchain: zephyr 0.11.1 (/opt/zephyr-sdk-0.11.1)
-- Found dtc: /opt/zephyr-sdk-0.11.1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/anpu/codedev/zephyrproject/zephyr/boards/arm/nrf52833dk_nrf52833/nrf52833dk_nrf52833.dts
-- Found devicetree overlay: /home/anpu/codedev/zephyrproject/bootloader/mcuboot/boot/zephyr/dts.overlay
-- Generated zephyr.dts: /home/anpu/codedev/zephyrproject/build/mcuboot_nrf52833/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/anpu/codedev/zephyrproject/build/mcuboot_nrf52833/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/anpu/codedev/zephyrproject/build/mcuboot_nrf52833/zephyr/include/generated/device_extern.h
Parsing /home/anpu/codedev/zephyrproject/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration '/home/anpu/codedev/zephyrproject/zephyr/boards/arm/nrf52833dk_nrf52833/nrf52833dk_nrf52833_defconfig'
Merged configuration '/home/anpu/codedev/zephyrproject/bootloader/mcuboot/boot/zephyr/prj.conf'
Configuration saved to '/home/anpu/codedev/zephyrproject/build/mcuboot_nrf52833/zephyr/.config'
Kconfig header saved to '/home/anpu/codedev/zephyrproject/build/mcuboot_nrf52833/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk-0.11.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Warning at /home/anpu/codedev/zephyrproject/zephyr/kernel/CMakeLists.txt:59 (message):
  Single threaded mode (CONFIG_MULTITHREADING=n) is deprecated

MCUBoot bootloader key file: /home/anpu/codedev/zephyrproject/bootloader/mcuboot/root-rsa-2048.pem
-- Configuring done
-- Generating done
-- Build files have been written to: /home/anpu/codedev/zephyrproject/build/mcuboot_nrf52833
-- west build: building application
[1/250] Preparing syscall dependency handling

[243/250] Linking C executable zephyr/zephyr_prebuilt.elf

[250/250] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       38684 B        48 KB     78.70%
            SRAM:       24064 B       128 KB     18.36%
        IDT_LIST:          0 GB         2 KB      0.00%
```

mjahamel, 2021-03-30:
This is strange, it consistently fails for me. Its clearly related to commit a7f9e9f "bootutil: Remove duplicated code from bootutil_misc.c" from 5 days ago, because if I go back to the commit before that I can build just fine. Its as though the change to bootutil-public.c to remove the static keywords from boot_write_trailer() & boot_write_trailer_flag() hasn't taken effect, yet I can see its there in the source. I'm doing a clean build as far as I can tell -- definitely deleting the build directory. Its as though the headers have been cached somewhere else. I'm using Zephyr-SDK 12.3, so I have a later compiler version, but I can't see what difference that would make. 

Corresponding log is:

michael@michael-ryzen5600:/company/zephyr/zephyr$ west build -p -b nrf52833dk_nrf52833 ../../mcuboot/boot/zephyr
-- west build: making build dir /company/zephyr/zephyr/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /company/zephyr/zephyr/cmake/app/boilerplate.cmake
-- Application: /company/mcuboot/boot/zephyr
-- Zephyr version: 2.5.99 (/company/zephyr/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: nrf52833dk_nrf52833
-- Cache files will be written to: /home/michael/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.12.3 (/home/michael/zephyr-sdk-0.12.3)
-- Found dtc: /home/michael/zephyr-sdk-0.12.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /company/zephyr/zephyr/boards/arm/nrf52833dk_nrf52833/nrf52833dk_nrf52833.dts
-- Found devicetree overlay: /company/mcuboot/boot/zephyr/dts.overlay
-- Generated zephyr.dts: /company/zephyr/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /company/zephyr/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /company/zephyr/zephyr/build/zephyr/include/generated/device_extern.h
Parsing /company/mcuboot/boot/zephyr/Kconfig
Loaded configuration '/company/zephyr/zephyr/boards/arm/nrf52833dk_nrf52833/nrf52833dk_nrf52833_defconfig'
Merged configuration '/company/mcuboot/boot/zephyr/prj.conf'
Configuration saved to '/company/zephyr/zephyr/build/zephyr/.config'
Kconfig header saved to '/company/zephyr/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/michael/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Warning at /company/zephyr/zephyr/kernel/CMakeLists.txt:59 (message):
  Single threaded mode (CONFIG_MULTITHREADING=n) is deprecated

MCUBoot bootloader key file: /company/mcuboot/root-rsa-2048.pem
-- Configuring done
-- Generating done
-- Build files have been written to: /company/zephyr/zephyr/build
-- west build: building application
[1/250] Preparing syscall dependency handling

[243/250] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map 
: && ccache /home/michael/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc    zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=/company/zephyr/zephyr/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/mpu/libarch__arm__core__aarch32__cortex_m__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/serial/libdrivers__serial.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  modules/mbedtls/lib..__modules__crypto__mbedtls.a  modules/mcuboot/lib..__bootloader__mcuboot__boot__bootutil__zephyr.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj  -L"/home/michael/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/thumb/v7e-m/nofp"  -L/company/zephyr/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /company/zephyr/zephyr/build/zephyr && /usr/bin/cmake -E echo
/home/michael/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: app/libapp.a(bootutil_misc.c.obj): in function `boot_write_copy_done':
/company/mcuboot/boot/bootutil/src/bootutil_misc.c:306: undefined reference to `boot_write_trailer_flag'
/home/michael/zephyr-sdk-0.12.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld: app/libapp.a(bootutil_misc.c.obj): in function `boot_write_swap_size':
/company/mcuboot/boot/bootutil/src/bootutil_misc.c:318: undefined reference to `boot_write_trailer'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /company/zephyr/zephyr/build

mjahamel, 2021-03-31:
It seems there was something wrong with my Zephyr install: a fresh install has fixed the problem. My apologies for the false alarm.

closed 2021-03-31 by mjahamel

GaneshMali-eaton, 2024-06-13:
I ran into the same problem, I think zephyr SDK and MCUboot compatibility issue. I am using zephyr 3.5 and SDK 0.16.5 and latest MCU I cloned. My friend is using SDK 0.16.1 and its working for him. 
Here is the error message:

←[92m-- west build: building application
[2/274] Generating include/generated/version.h
-- Zephyr version: 3.5.0 (C:/zephyr-v3.5.0/zephyr), build: zephyr-v3.5.0
[269/274] Linking C executable zephyr\zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map C:/mcuboot/mcuboot2/mcuboot1/boot/zephyr/build/zephyr/zephyr_pre0.map
C:\WINDOWS\system32\cmd.exe /C "cd . && C:\Users\e0784167\zephyr-sdk-0.16.5-1\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr\zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=C://mcuboot/mcuboot2/mcuboot1/boot/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/arch/arch/arm/core/cortex_m/cmse/libarch__arm__core__cortex_m__cmse.a  zephyr/arch/arch/arm/core/mpu/libarch__arm__core__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/soc/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/reset/libdrivers__reset.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/mcuboot/libmcuboot_util.a  modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  modules/mbedtls/libmbedTLSBase.a  modules/mbedtls/libmbedTLSCrypto.a  modules/mbedtls/libmbedTLSX509.a  modules/mbedtls/libmodules__mbedtls.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"c:/users/e0784167/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v8-m.main/nofp"  -LC://mcuboot/mcuboot2/mcuboot1/boot/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  modules/mbedtls/libmbedTLSBase.a  modules/mbedtls/libmbedTLSCrypto.a  modules/mbedtls/libmbedTLSX509.a  -mcpu=cortex-m33  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie && C:\WINDOWS\system32\cmd.exe /C "cd /D C:\\mcuboot\mcuboot2\mcuboot1\boot\zephyr\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E true""
c:/users/e0784167/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: app/libapp.a(swap_misc.c.obj): in function `swap_set_copy_done':
C:/mcuboot/mcuboot2/mcuboot1/boot/bootutil/src/swap_misc.c:200: undefined reference to `boot_write_copy_done'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\mcuboot\mcuboot2\mcuboot1\boot\zephyr\build'