mcu-tools/mcuboot #977: the mcuboot code(bin) be placed in a non-flash location ???

hongshui3000, 2021-04-13:
Can the mcuboot code be placed in a non-flash location? I use zephyr to develop and hope to solidify the mcuboot code into the rom (not flash) of my microcontroller

nvlsianpu, 2021-04-13:
Yes. MCUboot own location is something independent from application images location. MCUboot doesn't write anything to its location. Se no obstacles.

hongshui3000, 2021-04-14:
@nvlsianpu 
Thanks for the response

[image]
From the documentation, "boot_partition: for MCUboot itself" seems to have to be defined. I don't want to put it on the flash partition.

In the other way, I only want to place part of the mcuboot (Bluetooth OTA part) on the flash partition, whether it can perceive and integrate well with the mcuboot part of my rom

closed 2021-06-29 by github-actions[bot]