mcu-tools/mcuboot #978: switch cbor encoding to generated by cddl-gen

nvlsianpu, 2021-04-13:
Serial recovery mcumgr protocol implementation uses parser generated by cddl-gen while encoding uses tinycbot.
Since a while cddl-gen supports encoder generation which can be leveraged.

Expected results is flash footpint reduction and remove dependence from tinycbor library.

https://github.com/NordicSemiconductor/cddl-gen

nvlsianpu, 2021-04-13:
@oyvindronningstad Can you take this?

oyvindronningstad, 2021-04-14:
Yes, I've been looking into it already.

carlescufi, 2021-04-15:
> Yes, I've been looking into it already. You can assign me, I don't have permissions to do it myself :).

I just assigned you, thanks @oyvindronningstad.

de-nordic, 2021-05-10:
In progress here: https://github.com/mcu-tools/mcuboot/pull/997

commit 9f4aefd5 (2021-05-25): boot_serial: Port encoding to use cddl-gen

Non-generated. Using cbor_encode directly
This removes the dependence on TinyCBOR.

Fixes #978
closed 2021-05-25 by utzig