mcu-tools/mcuboot #983: Swap copy size optimization does not work if scratch area is larger than the image

boraozgen, 2021-04-26:
In the design document it is mentioned that the number of swapped sectors is optimized based on the image sizes in case the slot size is larger than the images:

> Note2: The bootloader tries to copy only used sectors (based on largest image installed on any of the slots), minimizing the amount of sectors copied and reducing the amount of time required for a swap operation.

This optimization does not work if the scratch area size is larger than the image size. `boot_copy_sz` function uses the scratch area size as the lower limit of its output. This output value is used without checking if the image size is smaller than the scratch area, resulting in a unnecessarily large copy operation.

As I was working on a patch, I realized that there is a larger logic error in the `swap_run` function. The last sector ID variables are incremented together, which means in case sector sizes of the primary and secondary slots are different, the slot with the larger sector size may get incremented to a point where the ID is more than the available sectors. Afterwards this ID is used to calculate the copy size in `boot_copy_sz`. Since the output is capped by the scratch area size, it does not matter but it is still a logic error.

My proposal would be to compare the output of `boot_copy_sz` with the largest of `primary_slot_size` and `secondary_slot_size`, and use the smallest number of this comparison. I could work on a PR if this issue is acknowledged.

hjuul, 2021-08-13:
I've got a problem that is likely to be related to the "logic error" you're mentioning.

- My firmware image is larger than the swap area size (swap is 256KiB).
- My internal flash has 256KiB sectors, and slot1 is in a QSPI flash with 4KiB sectors.

What I'm seeing is that MCUboot always swaps all sectors regardless of my image size. The swap is functional, but takes several minutes more than it should.

Slack discussion:
https://mcuboot.slack.com/archives/C6CB5BXGC/p1627994014013400

nvlsianpu, 2021-09-07:
> This optimization does not work if the scratch area size is larger than the image size. boot_copy_sz function uses the scratch area size as the lower limit of its output. This output value is used [...]

I beat that for-lop counting of the copy size will end execution if `last_sector_idx` decrements below zero. So ther is another limit which implies from image size to be copied.

closed 2022-03-21 by github-actions[bot]