mcu-tools/mcuboot #992: deltupdate supported?

iot-lorawan, 2021-05-06:
Hi,
    is there deltupdate feature supported?

utzig, 2021-05-06:
> is there deltupdate feature supported?

No.

d3zd3z, 2021-05-10:
Delta updates would probably also not be implemented in MCUboot itself. By the time we see the image in the second slot, it is already completed and ready to implemented.

The part of the application that downloads images and places them in this slot are where I would expect to find any delta update code. The application would download the compressed/delta image, while having access to the source image in the original slot (which will be needed to apply the delta update). Once the whole delta has been downloaded and applied, the target could then be rebooted, and MCUboot would perform the update.

closed 2021-05-12 by iot-lorawan