mcu-tools/mcuboot #1010: scratch section define in zephyr sysflash.h inappropriate

Cherish-forever, 2021-05-14:
In file zephyr/include/sysflash.h line 42:

`#if !defined(CONFIG_BOOT_SWAP_USING_MOVE)
#define FLASH_AREA_IMAGE_SCRATCH    FLASH_AREA_ID(image_scratch)
#endif`

that means:
CONFIG_BOOT_DIRECT_XIP=y or CONFIG_BOOT_UPGRADE_ONLY=y or CONFIG_BOOT_BOOTSTRAP=y
all those config must have scratch, I think CONFIG_BOOT_UPGRADE_ONLY and CONFIG_BOOT_DIRECT_XIP no need scratch section.

d3zd3z, 2021-05-17:
This suggests we always test with a partition table containing scratch (which makes a little bit of sense, since the test configurations can then also cover the situations with scratch.

This makes sense to test in the simulator.

utzig, 2021-05-17:
This issue was also seen on #1004.