mcu-tools/mcuboot #1017: The wrong elem_sz is calculated in `boot_write_sz` which may cause image swap may failure

Sherryzhang2, 2021-05-26:
In the API `boot_write_sz`, it chooses the bigger align value of the primary slot and the scratch area. For example, if the align value of the primary slot is 8 bytes while align value of the scratch area is 6 bytes, then the chosen value is 8 bytes. Then when writing the scratch area, mis-aligning issue probably happen which can cause write failure.

Instead of choosing the bigger one, I think the least common multiple of the two align values should be chosen.

utzig, 2021-05-26:
> For example, if the align value of the primary slot is 8 bytes while align value of the scratch area is 6 bytes, then the chosen value is 8 bytes.

It makes sense but do you know any real HW that has an align value which is not a power of two? What device uses 6 bytes?

Sherryzhang2, 2021-05-27:
Sorry I have no knowledge about the real HW. If normally the align value is always to be a power of 2, I agree to close this issue.

d3zd3z, 2021-05-28:
I can't imagine anyone building hardware where the alignment is not a power of two, as the address lines are generally just split off to make the distinction. Regardless, there are other places in the code that assume that the alignment is a power of two.

Maybe it'd be worth checking for this somewhere, and aborting really early if it is configured confusingly.

closed 2021-08-24 by d3zd3z