mcu-tools/mcuboot #1020: ESP32 support

qati, 2021-05-31:
Hi,

I'm trying to make mcuboot/zephyr work with ESP32. I've built the following dst for mcuboot:

```
&flash0 {
	partitions {
        compatible = "fixed-partitions";
        #address-cells = <1>;
        #size-cells = <1>;

        boot_partition: partition@0 {
            label = "mcuboot";
            reg = <0x000000000 0x00010000>;
        };
        slot0_partition: partition@10000 {
            label = "image-0";
            reg = <0x000010000 0x0007D000>;
        };
        slot1_partition: partition@0008D000 {
            label = "image-1";
            reg = <0x0008D000 0x0007D000>;
        };
        scratch_partition: partition@0010A000 {
            label = "image-scratch";
            reg = <0x0010A000 0x0007D000>;
        };
        storage_partition: partition@00187000 {
            label = "storage";
            reg = <0x00187000 0x0001F400>;
        };
	};
};
```

But when I am trying to build I get the following error:
```
/~zephyrproject/bootloader/mcuboot/boot/zephyr/include/target.h:42:2: error: #error "Target support is incomplete; cannot build mcuboot."
 #error "Target support is incomplete; cannot build mcuboot."
  ^~~~~
~/zephyrproject/bootloader/mcuboot/boot/zephyr/flash_map_extended.c:27:2: error: #error "FLASH_DEVICE_ID could not be determined"
 #error "FLASH_DEVICE_ID could not be determined"
  ^~~~~
~/zephyrproject/bootloader/mcuboot/boot/zephyr/flash_map_extended.c: In function 'flash_device_base':
~/zephyrproject/bootloader/mcuboot/boot/zephyr/flash_map_extended.c:42:18: error: 'FLASH_DEVICE_ID' undeclared (first use in this function); did you mean 'FLASH_AREA_ID'?
     if (fd_id != FLASH_DEVICE_ID) {
                  ^~~~~~~~~~~~~~~
                  FLASH_AREA_ID
```

I am not sure about this issue. Can somebody give some guidance?

sylvioalves, 2021-05-31:
@qati Hi Attila, we have been working on mcuboot support already. Unfortunately, only updating the DTS file is not enough as our current 2nd stage bootloader comes from IDF.  We will soon submit initial support to it.

qati, 2021-05-31:
@sylvioalves  Hi Sylvio, that sounds amazing. Looking forward for the PR. This support will make UpdateHub embedded Firmware Over-The-Air sample work right? Or something else is needed that as well?

qati, 2021-05-31:
Also will this integration make Non-Volatile Storage from zephyr available on esp32?

sylvioalves, 2021-06-01:
@qati, no, this initial PR won't allow OTA operation yet. It will take some weeks/months to get OTA feature.
Regarding NVS, it's been already supported for a while.

rftafas, 2021-06-22:
@qati PTAL: https://github.com/mcu-tools/mcuboot/pull/1032

closed 2022-01-03 by github-actions[bot]