mcu-tools/mcuboot #1034: Build time error when RAM_LOAD is enabled while MCUBOOT_DIRECT_XIP_REVERT is also enabled.

Sherryzhang2, 2021-06-23:
Build time error when RAM_LOAD is enabled while MCUBOOT_DIRECT_XIP_REVERT is also enabled. The reason is that in the context_boot_go() API which handles the RAM_LOADING and the DIRECT_XIP mode, it enables the code of supporting revert in XIP only by checking the macro /* MCUBOOT_DIRECT_XIP_REVERT */. But to make it better, I think /* MCUBOOT_DIRECT_XIP_REVERT */. should only take effect in DIRECT_XIP mode. Otherwise, build time error occur if the user selects RAM_LOADING and also enables the MCUBOOT_DIRECT_XIP_REVERT.

d3zd3z, 2021-06-25:
I'll look at this after #1019 merges. This tests these features in the simulator, so I can make sure this combination is both buildable, as well as have a test for it.

davidvincze, 2021-06-28:
As I see the build time error is caused by a misconfiguration. The fastest solution could be either applying `#if defined(MCUBOOT_DIRECT_XIP) && defined(MCUBOOT_DIRECT_XIP_REVERT)` in the right places or adding the corresponding configuration check to 'bootutil_priv.h' as it already contains a few. @Sherryzhang2  what's your view on it?
 
Long term solution: start supporting revert mechanism with the RAM_LOAD strategy as well?

d3zd3z, 2021-06-28:
> As I see the build time error is caused by a misconfiguration.

I believe I've fixed this in one of the commits in #1019.

d3zd3z, 2021-06-28:
Actually, maybe not, I think I fixed a different configuration combination issue.

Sherryzhang2, 2021-06-29:
> As I see the build time error is caused by a misconfiguration. The fastest solution could be either applying `#if defined(MCUBOOT_DIRECT_XIP) && defined(MCUBOOT_DIRECT_XIP_REVERT)` in the right [...]

 I prefer "adding #if defined(MCUBOOT_DIRECT_XIP) && defined(MCUBOOT_DIRECT_XIP_REVERT)".

davidvincze, 2021-06-29:
@Sherryzhang2, any comments? : https://github.com/mcu-tools/mcuboot/pull/1042

Sherryzhang2, 2021-07-20:
I think this issue can be closed now.

closed 2021-07-20 by Sherryzhang2