mcu-tools/mcuboot #1043: "ModuleNotFoundError: No module named 'edtlib'" with Zephyr 2.6.0

endiancj, 2021-06-30:
According to this PR in Zephyr [zephyrproject-rtos#33746](https://github.com/zephyrproject-rtos/zephyr/pull/33746) it seems the devicetree pip package is being made a standalone project to be used with other projects than only zephyr. In this work  edtlib.py and a few other python files have been moved to a new directory. This breaks the build of mcuboot together with zephyr 2.6.0.

Old path: `zephyr/scripts/dts/edtlib.py`
New path: `zephyr/scripts/dts/python-devicetree/src/devicetree/edtlib.py`

Because of the new path of edtlib.py `mcuboot/scripts/assemble.py` can't find it when trying to import it.

`assemble.py`  needs to be updated with this new path of edtlib.py

nvlsianpu, 2021-06-30:
Strange that I didn't encounter this issue. Thanks for reporting!

endiancj, 2021-07-26:
closed the issue since the PR  [Update path to devicetree python package lib files #1044](https://github.com/mcu-tools/mcuboot/pull/1044) resolved it.

closed 2021-07-26 by endiancj