mcu-tools/mcuboot #1048: Support new mbedtls library

Sherryzhang2, 2021-07-05:
TFM has the requirement of upgrading the MBEDTLS library from mbedtls-2.26.0 to the coming v3.0. Some APIs that are used in MCUboot are renamed according to https://github.com/ARMmbed/mbedtls/blob/development/docs/3.0-migration-guide.md#rename-mbedtls__ret-cryptography[…]ecated-variants-have-been-removed . So build time error will occur in MCUboot with nearly v3.0 MBEDTLS.
So we propose

- Align the mbedtls APIs used in mcuboot\boot\bootutil\include\bootutil\crypto to the coming mbedtls v3.0

- Also, provide backward compatibility by defining bootutil_xxx to different version of mbedtls API.

d3zd3z, 2021-07-07:
Zephyr will likely migrate to mbedtls 3.0. If we can get the timing right, we might just be able to migrate to the new apis entirely.

Sherryzhang2, 2021-07-08:
TFM code freeze for the next release is planned to happen in the middle of July. So what's the status of the migration work currently? Is it just a plan or it has been in progress?

Sherryzhang2, 2021-07-08:
I created a PR for this issue. https://github.com/mcu-tools/mcuboot/pull/1051

nvlsianpu, 2021-07-20:
@Sherryzhang2 Can this be closed?

Sherryzhang2, 2021-07-20:
Yes. I think it can be closed now.

closed 2021-07-20 by Sherryzhang2