mcu-tools/mcuboot #1066: boot_set_confirmed not working as expected

vleeb, 2021-07-16:
hi everybody,
i am kind of a beginner with mcuboot with zephyr and started to go through the "hello world" sample on a nrf52840dk board.
i got it working to build/flash the mcuboot and the first hello world application into the primary image, giving the following output:

```
*** Booting Zephyr OS build zephyr-v2.6.0-440-gc2068e5d6e87  ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Swap type: none
I: Bootloader chainload address offset: 0xc000
I: Jumping to the first image slot
*** Booting Zephyr OS build zephyr-v2.6.0-440-gc2068e5d6e87  ***
Hello World from alfa on nrf52840dk_nrf52840!
```
then i tried to modify the main.c to confirm the image in the primary slot with the following code:
```
  if ( boot_is_img_confirmed() )
  {
    printk("image is confirmed\n" );
  }
  else
  {
    printk("image is not confirmed\n" );
    printk("confirming now...\n" );
    int rc = boot_write_img_confirmed();
    if ( rc == 0 )
    {
      printk("...successfully confirmed\n" );
    }
    else
    {
      printk("...failed to confirm\n" );
    }
    // check if actually confirmed now
    if ( boot_is_img_confirmed() )
    {
      printk("image is confirmed now\n" );
    }
    else
    {
      printk("image is not confirmed now\n" );
    }
  }
```

getting the following output:
```
*** Booting Zephyr OS build zephyr-v2.6.0-440-gc2068e5d6e87  ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Swap type: none
I: Bootloader chainload address offset: 0xc000
I: Jumping to the first image slot
*** Booting Zephyr OS build zephyr-v2.6.0-440-gc2068e5d6e87  ***
image is not confirmed
confirming now...
...successfully confirmed
image is not confirmed now
```
... leaving me quite confused.

with debugging i found the function boot_set_confirmed_multi(...) in file bootutil_public.c, doing something unexpected in the switch:
```
switch (state_primary_slot.magic) {
    case BOOT_MAGIC_GOOD:
        /* Confirm needed; proceed. */
        break;

    case BOOT_MAGIC_UNSET:
        /* Already confirmed. */
        goto done;

    case BOOT_MAGIC_BAD:
        /* Unexpected state. */
        rc = BOOT_EBADVECT;
        goto done;
    }
```
state_primary_slot.magic is equal to BOOT_MAGIC_UNSET ... and therefore treated as "already confirmed" and doing nothing.

so do i use it in the wrong way? 
or is there a problem in the function boot_is_img_confirmed()? 
or is there a problem in the function boot_set_confirmed_multi()?

d3zd3z, 2021-07-16:
At least the snippet above does look like a bug. I haven't used this code myself, so that may be why I've missed this. It might be useful to add tests of this functionality to the simulator so we can at least verify they work.

vleeb, 2021-07-19:
Hi David,

thank you for your reply. Is there anything I can do to support?

Some more observations from my trials:
I modified the above switch/case to treat BOOT_MAGIC_UNSET as "confirm needed". Then the confirmation actually happens and it is then also stored persistently.

nvlsianpu, 2021-07-19:
@d3zd3z  Dosen't BOOT_MAGIC_UNSET support image directly flashed to SoC primary slot? Such image doesn't need to be confirmed.

I think this is not a buggy behavior.

d3zd3z, 2021-07-21:
@nvlsianpu If it is flashed directly into the primary slot, it should already be marked. The marking is intended for the first boot of an upgrade. A subsequent boot would detect this, and do the revert. It sounds like this might just be that first case where the image was directly flashed.

vleeb, 2021-07-22:
Hi David,
thank you for replying. 

Yes, you are right … I am talking about the very fist flashing of an image to the primary slot. As this version of the application software should behave like upcoming updates, it should also check, if it was marked … and should mark itself if not done already.

After having gained more insight into the behavior of mcu-boot, I would say that my problem is just a corner case, but I would still expect that boot_set_confirmed() should not return ok, if it is doing nothing. If it is used with an image, that is not intended to be marked, it should return some error.

Just an suggestion for an improvement.

Volker

nvlsianpu, 2021-07-22:
> @nvlsianpu If it is flashed directly into the primary slot, it should already be marked. The marking is intended for the first boot of an upgrade. A subsequent boot would detect this, and do the revert. It sounds like this might just be that first case where the image was directly flashed.

Yes this is about the early stage of device firmware, before any firmware upgrade happened, when primary image slot was flashed directly using a programing interface directly. MCUboot thread such image as confirmed. Code can be modified so it even will be marked as confirmed physically - but this won't change behavior.

> After having gained more insight into the behavior of mcu-boot, I would say that my problem is just a corner case, but I would still expect that boot_set_confirmed() should not return ok, if it is [...]

IMO the API must return success as firmware is working correctly, this is not a issue/exception that has to be served somehow - everything is fine.

BTW You might sign image with --pad option so you image will contain the magic in image status field.

d3zd3z, 2021-08-23:
Checking on the status of this issue. Is there something we'd like to fix?

vleeb, 2021-08-24:
Hi David,

 

for me it still seems to be wrong, when boot_set_confirmed() returns no-error, but does not do anything. And the next call to boot_is_img_confirmed() returns false again.

This is not a severe issue, but it would have helped me a lot at the beginning to understand that I was calling this function for an image what was not intended to be confirmed.

 

Andrzej does not think, that this is a wrong behavior.

 

So feel free to make an improvement or keep it like it is.

 

Thank you for your support.

 

Volker

d3zd3z, 2021-08-24:
Closing this, please reopen, or create another issue if needed.

closed 2021-08-24 by d3zd3z