mcu-tools/mcuboot #1068: zephyr mcuboot project using tinycrypt

ChrisKurz, 2021-07-20:
Hi, I am currently working on the mcuboot project. I want to use mcuboot and encrypt application image. So within the mcuboot I want to use TinyCrypt. I have some problems to get TinyCrypt enabled. My understanding is that there is an own tinycrypt version within the mcuboot project and CONFIG_BOOT_USE_TINYCRYPT enables it. But this is an invisible config symbol. What is the intended way to enable this library? My assumption is that I have to modify the default value in the KCONFIG file where CONFIG_BOOT_USE_TINYCRYPT is defined. Or is there a better way?
Regards, chris

nvlsianpu, 2021-07-20:
As I see this question is not to Mcuboot here but to the Nordic NCS fork.
CONFIG_BOOT_USE_TINYCRYPT is not selectable (is a hidden property).

TINYCRYPT is selected indirectly with signature type (some of them has more crypto backends to be selected).

d3zd3z, 2021-07-21:
It is possible to use EC256 with either Tinycrypt, or Mbed TLS. I'm not sure if the config files reflect this possibility, though. There are tradeoffs to each library. (tinycrypt is, well, tiny, but also slower than Mbed TLS for the EC operations).

d3zd3z, 2021-07-21:
To clarify a bit further. Each signing algorithm has one or more implementations that make use of that library. Currently, RSA always uses Mbed TLS, and EC256 can use either Mbed TLS or tinycrypt. The other signing algorithm, ed25519 has its own library to support it, in the tree.

ChrisKurz, 2021-07-23:
Hi David,
thanks for your feedback. 
Signature seems to work quite well, only enabling an encrypted Image causes some problems.
I took a closer look and it look into it.
For example, when selecting CONFIG_BOOT_ENCRYPT_EC256 the Tinycrypt is not automatically selected within its definition. compiling the project resulted into an error stating that either tinycrpyte of mbedtls is needed... 
There is the possibility to use CONFIG_BOOT_TINYCRPT to enable tinycrypt in case the signature CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256 is selected... 

I think I found a way how to continue with my project...

closed 2021-07-23 by ChrisKurz