mcu-tools/mcuboot #1085: Serial Recovery uploads to the slot 0 (instead of slot 1)

amateusz, 2021-08-02:
Serial recovery upload always  

I have custom nrf52840 board with (USB & Serial) and I'm trying to make MCUboot to work through both transports.
 
USB DFU firmware update flow works as in docs:
- `dfu-util -D build/zephyr/zephyr.signed.bin -a "image-1"`
- bootloader swaps and tests `Swap type: test`
- once confirmed, only the new image persists 

Images:
 image=0 slot=0
    version: 1.0.0.3
    bootable: false
    flags: 
    hash: Unavailable
Split status: N/A (0)

Images:
 image=0 slot=0
    version: 1.0.0.3
    bootable: false
    flags: 
    hash: Unavailable
 image=0 slot=1
    version: 1.0.0.4
    bootable: false
    flags: 
    hash: Unavailable
Split status: N/A (0)

Images:
 image=0 slot=0
    version: 1.0.0.4
    bootable: false
    flags: 
    hash: Unavailable

Serial Recovery always uploads to the slot 0:
(The new image here is 1.0.0.5)
```
Images:
 image=0 slot=0
    version: 1.0.0.3
    bootable: false
    flags: 
    hash: Unavailable
 image=0 slot=1
    version: 1.0.0.4
    bootable: false
    flags: 
    hash: Unavailable
Split status: N/A (0)
```
`$ ~/go/bin/mcumgr --timeout 500000 --conntype serial --connstring "dev=/dev/serial/by-id/usb-FTDI_FT232R_*****-port0,baud=115200" image upload -e zephyr/zephyr.signed.bin`

`$ ~/go/bin/mcumgr --timeout 500000 --conntype serial --connstring "dev=/dev/serial/by-id/usb-FTDI_FT232R_*****-port0,baud=115200" image list`

```
Images:
 image=0 slot=0
    version: 1.0.0.5
    bootable: false
    flags: 
    hash: Unavailable
 image=0 slot=1
    version: 1.0.0.4
    bootable: false
    flags: 
    hash: Unavailable
Split status: N/A (0)
```

de-nordic, 2021-08-06:
@amateusz The serial recovery is designed to always upload to primary slot, because it is recovery: it is used to fix your board when your primary application fails and the board, except for being able to run mcuboot, is paperweight. 
This behaviour can be altered with https://github.com/mcu-tools/mcuboot/commit/6c8932e9d044042cf882c62e83012081436d8cc9

In case of USB DFU I think that there is a bug; the USB DFU, when enabled for mcuboot, should always allow to upload to primary slot and alternatively to secondary; it makes no sense to upload to secondary slot, from the mcuboot, as  there is not app running, at this point from primary slot - this is similar to recovery scenario.

Uploading to secondary slot only makes sense when you have running application, because you can not overwrite the active application, which is in the primary slot.

de-nordic, 2021-08-06:
I have allowed myself to tag the title of your issue with "zephyr", as this is zephyr specific issue.

d3zd3z, 2021-08-12:
Is there a requirement to be able to serial recovery upload to slot 1? If so, this should be marked as an enhancement. Otherwise, I think this should be closed.

nvlsianpu, 2021-08-16:
Updating the primary slot with USB DFU class  (by default ) as mcuboot recovery method is intentional behavior.

d3zd3z, 2021-08-16:
I'll go ahead an close this issue, then. If there is a need to be able to write to slot 1 from serial recovery, it'd probably be best to open a new issue and explain there why that functionality is needed. We can then call it an enhancement, since the existing functionality is behaving as designed.

closed 2021-08-16 by d3zd3z