mcu-tools/mcuboot #1091: zephyr: mcuboot configured with CONFIG_BOOT_SIGNATURE_TYPE_NONE will not boot application

de-nordic, 2021-08-06:
MCUboot configuration:
```
CONFIG_MBEDTLS=n
CONFIG_BOOT_SIGNATURE_TYPE_NONE=y
```

Testing on hello application configured with:
```
CONFIG_BOOTLOADER_MCUBOOT
```

signed with commands:
```
imgtool sign --align 8 --slot-size  0x67000 --header-size 0x200 -v 0.0.0 zephyr.hex zephyr.signed2.hex
```

Uploaded to nrf52840dk board
```
nrfjprog --program zephyr.signed2.hex --sectorerase
```
gives console output:
```
*** Booting Zephyr OS build zephyr-v2.6.0-1773-gca5921845d42  ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Swap type: none
I: Bootloader chainload address offset: 0xc000
I: Jumping to the first image slot
```

It seems that application gets verified and allowed to boot, but nothing happens then.

Tested on:
mcuboot/main:  420ad9adbd644fd43516d0e59080a84935aa72de
zephyr/main: ca5921845d4241ce9da9496a8e97d7dd6c1c4bae

d3zd3z, 2021-11-24:
I just tried to recreate this with current main 5e762643778, and I am not able to reproduce it. Does this configuration work if you enable a signature type? It might be necessary to use the debugger on this, with your specific instance.

de-nordic, 2021-11-25:
> I just tried to recreate this with current main [5e76264](https://github.com/mcu-tools/mcuboot/commit/5e762643778454ca9a1f872e3dfdef6490204f96), and I am not able to reproduce it. Does this configuration work if you enable a signature type? It might be necessary to use the debugger on this, with your specific instance.

This is very old issue. I will check it, maybe it is gone.

de-nordic, 2022-03-21:
> @de-nordic Any update on this?

I am really sorry, I could not find time to check this. I will try to do the test Tomorrow.

utzig, 2022-03-30:
Just tried this on a mass erased board:

```
west build -b nrf52840dk_nrf52840 -d build_boot bootloader/mcuboot/boot/zephyr -- -DCONFIG_BOOT_SIGNATURE_TYPE_NONE=y
west flash -d build_boot

west build -b nrf52840dk_nrf52840 zephyr/samples/subsys/mgmt/mcumgr/smp_svr
imgtool sign --align 8 --slot-size 0x67000 --header-size 0x200 -v 0.0.0 build/zephyr/zephyr.hex zephyr.signed.hex
nrfjprog --program zephyr.signed.hex --sectorerase
```

It boots without issues; also works by using `west sign`, although I think it adds the signature in that case. I checked that the build is really using only hash, and in my case it is using Tinycrypt. If it was using MbedTLS then I would guess it could be some issue after the upgrade to 3.x...

utzig, 2022-03-30:
and `samples/hello_world` also worked.

nvlsianpu, 2022-03-30:
I can't reproduce the issue as well.

nvlsianpu, 2022-03-30:
Maybe the sample had issue wit logging at the time this ticket was registered,

I'm closing the ticket, fill free to re-open if needed,

closed 2022-03-30 by nvlsianpu