mcu-tools/mcuboot #1094: Consider not storing dev/test keys in repo

d3zd3z, 2021-08-09:
Currently, development keys are stored in the repo. As long as the instructions are followed, this should not result in these keys being used in production. However it is possible for this to happen. One way to fix this is to remove these keys, and instead, have all of the systems that used the development keys instead create a set if they are needed. Ideally, this would be created once in a developer's workspace, and the same keys would be reused. This needs to be coordinated across all of the systems that use MCUboot, including the simulator, Zephyr, MyNewt, and the other supported platforms.

bn-norbit, 2021-09-17:
#1088 can also address this issue for Zephyr. If the default keys are removed, the user is forced to generate an encryption key and set  CONFIG_BOOT_ENCRYPT_KEY_FILE accordingly. The encryption tests can generate a keys on the fly.

d3zd3z, 2021-09-18:
Thanks @bn-norbit, This does help this for Zephyr. A challenge with this issue is addressing it for every platform before the keys can be removed. But, maybe when we address it for a given platform, that platform needs to be changed so that it never uses the in-tree keys.

closed 2022-03-31 by github-actions[bot]