mcu-tools/mcuboot #1127: Probably incompatibility between Zephyr GIT and or cmake 3.21.1

holgerschurig, 2021-09-03:
Hi,

recently Zephyr upgraded the needed cmake version to a minimum of 3.20.0 (see https://github.com/zephyrproject-rtos/zephyr/commit/1cccc8a8feda1b1cbff4a81fbb552b75f0f1ce28).

So I updated my cmake, and went to the newest version, 3.21.1.

Despite some warnings, I can still compile my program for posix and nucleo_f303re board.

Now I tried to make mcuboot running on the nucleo_f303re board. I configure with

```
	cmake \
		-B build-mcuboot \
		-GNinja \
		-Wno-dev \
		-Wno-deprecated \
		-DBOARD=nucleo_f303re \
		-DDTC_OVERLAY_FILE="$(PWD)/nucleo_f303re.overlay" \
		$(PWD)/mcuboot/boot/zephyr
```
but later in the compilation progress, I see that the symbol MCUBOOT_BOOTUTIL hasn't been set, as there is a command line option `-lMCUBOOT_BOOTUTIL` in the link step:

```
$ ninja -C build-mcuboot
ninja: Entering directory `build-mcuboot'
[1/8] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map /home/schurig/d/v73/build-mcuboot/zephyr/zephyr_prebuilt.map 
: && ccache /usr/bin/arm-none-eabi-gcc   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr/zephyr_prebuilt.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=/home/schurig/d/v73/build-mcuboot/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/drivers/interrupt_controller/libdrivers__interrupt_controller.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/pinmux/libdrivers__pinmux.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/stm32/stm32cube/lib..__modules__hal__stm32__stm32cube.a  modules/mbedtls/libmodules__mbedtls.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/usr/lib/gcc/arm-none-eabi/7.3.1/thumb/v7e-m"  -L/home/schurig/d/v73/build-mcuboot/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -lMCUBOOT_BOOTUTIL && cd /home/schurig/d/v73/build-mcuboot/zephyr && /usr/bin/cmake -E echo
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld.bfd: cannot find -lMCUBOOT_BOOTUTIL
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

In the zephyr configuration, CONFIG_MCUBOOT_BOOTUTIL is however correctly defined:

```
$ grep BOOTU build-mcuboot/zephyr/.config 
CONFIG_MCUBOOT_BOOTUTIL_LIB=y
```

Zephyr and mcuboot are both at today's git HEAD:

zephyr: e12e126f3262ffcac90572a6be4dffe12f17a30f
mcuboot: ae35510b58016b9aa6dd1dff55239bba8acd45da

nvlsianpu, 2021-09-03:
I'm able to build mcuboot without any problem on given revisions (Of course for different board).
Worth to notice that I'm using zephyr-rtos `west build` command.

mlsvrts, 2021-11-08:
@vindicatorr This error:

```
mcuboot/boot/zephyr/serial_adapter.c: In function 'boot_uart_fifo_init':
<pathTo>/zephyr/include/sys/util.h:62:55: error: size of unnamed array is negative
   62 | #define ZERO_OR_COMPILE_ERROR(cond) ((int) sizeof(char[1 - 2 * !(cond)]) - 1)
      |                                                       ^
<pathTo>/zephyr/include/sys/util_internal.h:67:26: note: in definition of macro '__DEBRACKET'
   67 | #define __DEBRACKET(...) __VA_ARGS__
      |                          ^~~~~~~~~~~
<pathTo>/zephyr/include/sys/util_internal.h:59:9: note: in expansion of macro '__GET_ARG2_DEBRACKET'
   59 |         __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
      |         ^~~~~~~~~~~~~~~~~~~~
<pathTo>/zephyr/include/sys/util_internal.h:54:9: note: in expansion of macro '__COND_CODE'
   54 |         __COND_CODE(_XXXX##_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~
<pathTo>/zephyr/include/sys/util_macro.h:157:9: note: in expansion of macro 'Z_COND_CODE_1'
  157 |         Z_COND_CODE_1(_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~~~
<pathTo>/zephyr/include/device.h:300:9: note: in expansion of macro 'COND_CODE_1'
  300 |         COND_CODE_1(DT_HAS_COMPAT_STATUS_OKAY(compat),                      \
      |         ^~~~~~~~~~~
<pathTo>/zephyr/include/device.h:302:22: note: in expansion of macro 'ZERO_OR_COMPILE_ERROR'
  302 |                     (ZERO_OR_COMPILE_ERROR(0)))
      |                      ^~~~~~~~~~~~~~~~~~~~~
<pathTo>/mcuboot/boot/zephyr/serial_adapter.c:197:20: note: in expansion of macro 'DEVICE_DT_GET_ONE'
  197 |         uart_dev = DEVICE_DT_GET_ONE(zephyr_cdc_acm_uart);
      |                    ^~~~~~~~~~~~~~~~~
make[2]: *** [zephyr/CMakeFiles/zephyr.dir/build.make:551:

```

Is because, since Zephyr LTS2 (2.7.0), the cdc acm uart is defined by the device tree instead of the project configuration. Have you tried building with a dt overlay that enables the console? (See: https://github.com/mcu-tools/mcuboot/blob/main/boot/zephyr/usb_cdc_acm.overlay)

Of course, I am not sure that this addresses the overall pain-point of dependency mismatches when trying to build without `west`.

vindicatorr, 2021-11-08:
I'll certainly consider giving that a go at some point.
I pulled back a bit and sticking with the native SDKs for the moment, as I've encountered issues with esp32/zephyr and found that it's also WIP and I'm only just now starting to get a handle on cmake and taking a closer look at device-trees.

d3zd3z, 2021-11-15:
I believe west is now a hard requirement to build Zephyr. It doesn't just invoke cmake, but the Zephyr build system invokes west to find out other information. I'm a little surprised, though, that it just doesn't fail the build early with a check.

vindicatorr, 2021-11-15:
Until I see https://docs.zephyrproject.org/latest/guides/west/without-west.html get revoked/deprecated, I won't be seeing west as a "hard requirement".

nvlsianpu, 2021-11-16:
@vindicatorr Your logs are looking like result of build old project with newer zephyr-rtos version version build issue. Did you update zephyr west modules .Did you try fresh build?

Lets-a-go, 2022-01-08:
Fresh build on fresh system on pristine drive still fails, I can attest. I am on yet another board (nrf52840dongle_52840, and am trying others now--they all hit the same problem so I don't believe this is an isolated incident!). It's quite broken, and only about 5 or so cut-and-paste steps in from the "Getting started" point for this board as far as west/zephyr go. DT overlay comment above is helpful (thanks @mlsvrts) If I come up with an easy way to force a sync, I'll post back.  I'm not a DT expert, but I'll see if I can dig something up.

d3zd3z, 2022-01-11:
An additional data point, I have similar build failures on the lpcexpresso55s69 when using cmake instead of west. It may be specific issues on various targets combined with most users using west and not seeing these kinds of issues.

closed 2022-07-25 by github-actions[bot]