mcu-tools/mcuboot #1128: zephyr: Unable to flash signed (ZMK) image to Particle Xenon (nRF52840)

dzervas, 2021-09-05:
hey, I'm having a hard time making MCUBoot working on my Particle Xenon (nrf52840). Data:
- Using https://github.com/zephyrproject-rtos/mcuboot tag v1.7.2
- defconfig: https://github.com/dzervas/zmk-config/blob/f1f4311d7a13683f1759ec1a78ca311bffe6b647/config/boards/shields/lab68/lab68_mcuboot_defconfig
- overlay (on top of upstream zephyr particle xenon overlay): https://github.com/dzervas/zmk-config/blob/f1f4311d7a13683f1759ec1a78ca311bffe6b647/config/boards/shields/lab68/lab68_mcuboot.overlay
- mcuboot compiles & flashes good (using openocd, as described in https://github.com/dzervas/zmk-config/blob/f1f4311d7a13683f1759ec1a78ca311bffe6b647/README.md#build--flash-mcuboot)

Problems are:
- I can't pull any MCUBoot logs even though I've log enabled in defconfig (I connect with serial but I see nothing and no response)
- when I flash the signed ZMK image (I verified it and the key is correct) mcumgr image list remains empty

```
$ imgtool verify -k mcuboot/signing-key-ed25519.pem zmk.signed.bin
Image was correctly validated
Image version: 0.0.0+0
Image digest: a24e0ddb2f0bfc84e273c6c508bf2c694ecbb6d0e1b42693c66a89d08a39af2c
$ mcumgr --conntype=serial --connstring='dev=/dev/ttyACM0,baud=115200' image upload -e zmk.signed.bin
 265.31 KiB / 265.31 KiB [=============================================================================================================================================================================================] 100.00% 1.07 KiB/s 4m8s
Done
$ mcumgr --conntype=serial --connstring='dev=/dev/ttyACM0,baud=115200' image list
Images:
 image=0 slot=0
    version: 0.0.0.0
    bootable: false
    flags:
    hash: Unavailable
Split status: N/A (0)
```

nvlsianpu, 2021-09-06:
@dzervas The Application should be build with same partitions arrangement as the bootloader. I didn't see this in commanding you pointed.

dzervas, 2021-09-06:
The app is built using the following overlay (which includes the mcuboot used overlay): https://github.com/dzervas/zmk-config/blob/f1f4311d7a13683f1759ec1a78ca311bffe6b647/config/boards/shields/lab68/lab68.overlay

nvlsianpu, 2021-09-07:
@dzervas Which vesrion of the zephyr-project do you using - seams mcuboot 1.7.0-branch is incompatible with zephyr-rtos mainline since a while.

dzervas, 2021-09-07:
I'm building (almost) mainline ZMK which uses zephyr https://github.com/zmkfirmware/zephyr/tree/v2.5.0+zmk-fixes

dzervas, 2021-09-09:
also apparently it is flashed correctly and it runs as it is. but why can't I see the hash and it shows as not bootable? maybe I have to set the version?

nvlsianpu, 2021-09-09:
serial recovery doesn't support SHA presentation. Regard the flag and the bootable I don't remember.

dzervas, 2021-09-10:
oooooh that makes sense then!
Apparently the bootable flag is shown wrong too as the image works correctly
Maybe I should use DFU? (it's a different protocol, right?)

closed 2022-03-24 by github-actions[bot]