mcu-tools/mcuboot #1146: boot/bootutil: image header size check is not precise enough

nvlsianpu, 2021-09-16:
related to #1140 

looks like `boot_is_header_valid()` is a very rough check regard sizes: it checks whether the header and image binaries fit in the slot.
Slot must be able to accommodate other entities like TLVs, the image trailer and the swap status.

I'm not sure whether this check shouldn't be more precise.

nvlsianpu, 2021-09-16:
@d3zd3z Are you answer to the questions I'm asking above?

d3zd3z, 2021-09-17:
Provoked by #1140, I've started trying to add size limit checking to the simulator. What I want to do is be able to make an image that is exactly the largest size possible, and make sure it works.

But, yes, the maximum size does need to include the header, the image itself, the TLV, status, and the trailer. At best, they can all be packed together for a solid image.

The TLV is the trickiest, as it changes size depend on the things that are included within it.

utzig, 2021-10-07:
> Actually, it might fail pretty badly if swap ends up blowing away the end of the TLV in the process of writing the trailer.

This will probably happen.

nvlsianpu, 2022-01-21:
looks like bootutil_img_validate() is a place where precise check can be added, alternatively the check can be added somewhere around.

utzig, 2022-01-21:
> looks like bootutil_img_validate() is a place where precise check can be added, alternatively the check can be added somewhere around.

I don't disagree (have not looked at the code!), but if there is a problem with the checks, maybe add some simulator test that exercises the incorrect behavior?

nvlsianpu, 2022-01-21:
I'm going to provide a patch for this ticket, will try to add test if I manage to.

utzig, 2022-08-23:
Fixed by #1286 (hopefully the bot won't be closing this again)

closed 2023-03-06 by github-actions[bot]

nvlsianpu, 2023-03-06:
> Fixed by https://github.com/mcu-tools/mcuboot/pull/1286 (hopefully the bot won't be closing this again)

Yes, It was fixed.