mcu-tools/mcuboot #1148: ModuleNotFoundError: No module named 'edtlib'

maweicoltsmart, 2021-09-17:
zephyrproject/bootloader/mcuboot/samples/zephyr

`make BOARD=frdm_k64f full.bin`

```
Traceback (most recent call last):
  File "../../scripts/assemble.py", line 148, in <module>
    main()
  File "../../scripts/assemble.py", line 131, in main
    import edtlib
ModuleNotFoundError: No module named 'edtlib'
Makefile:94: recipe for target 'full.bin' failed

```

nvlsianpu, 2021-09-17:
I think this is already resolved, see https://github.com/mcu-tools/mcuboot/issues/1043
Did you use an old repo version?

maweicoltsmart, 2021-09-17:
Yes, I updated the code, no errors were reported, but another strange question appeared: Why is the size of the two mirrors so different in the same program?

```
../../scripts/imgtool.py sign \
	--key ../../root-rsa-2048.pem \
	--header-size 0x200 \
	--align 8 \
	--version 1.2 \
	--slot-size 0x60000 \
	--pad \
	/home/colt/share/zephyrproject/code/zephyrproject/bootloader/mcuboot/samples/zephyr/build/wch_eth_ch579/hello2/zephyr/zephyr.bin \
	signed-hello2.bin
../../scripts/assemble.py -b /home/colt/share/zephyrproject/code/zephyrproject/bootloader/mcuboot/samples/zephyr/build/wch_eth_ch579/mcuboot \
    -z /home/colt/share/zephyrproject/code/zephyrproject/zephyr \
    -p signed-hello1.bin \
    -s signed-hello2.bin \
    -o full.bin
partition mcuboot, pos=0, offset=0
partition image-0, pos=41784, offset=65536
partition image-1, pos=117184, offset=156672
Traceback (most recent call last):
  File "../../scripts/assemble.py", line 148, in <module>
    main()
  File "../../scripts/assemble.py", line 145, in main
    output.add_image(args.secondary, "image-1")
  File "../../scripts/assemble.py", line 94, in add_image
    raise Exception("Image {} is too large for partition".format(source))
Exception: Image signed-hello2.bin is too large for partition
Makefile:94: recipe for target 'full.bin' failed
make: *** [full.bin] Error 1

```

closed 2021-09-25 by maweicoltsmart