mcu-tools/mcuboot #1151: Encode version information into MCUboot binary

d3zd3z, 2021-09-17:
Encode the version information about the MCUboot code used for a build into the binary image. This can be used to print the version information, and also used to find this information in a binary.

This will likely have to be done in different ways, depending on the build environment.

- [ ] Zephyr
- [ ] Mynewt
- [ ] Cypress
- [ ] Espressif
- [ ] Mbed
- [ ] Nuttx

Overall, the build system should aim to set a defined variable, say `BUILD_VERSION` that can be expanded in the code. Or if this is infeasible in any of the platforms, this should be defined by the platform file to provide this information.

Cherish-Gww, 2021-11-17:
Is there any new progress?

d3zd3z, 2021-11-18:
I wouldn't expect any progress, as I'm not aware of anyone working on this.

closed 2022-06-01 by github-actions[bot]

julian-yeoh-setec, 2023-06-14:
Hi! I am also interested in this feature. Has this feature been implemented? If so, can someone please point me to the steps for obtaining the version of MCUBOOT from the bootloader binary (I would like to query the bootloader version from the application). Thanks.