mcu-tools/mcuboot #1154: Decouple FI test from TF-M Secure executable

tamasban, 2021-09-21:
Fault Injection test relies on TF-M repository:
- build and link the AN521 BSP code to MCUboot
- Secure and Non-secure images are executed in the QEMU environment

The goal of the test is to determine whether the system is capable of boot-up properly with a tampered image (a signature tampers).  The test greps for a string that is printed by TF-M secure image. Any change in this message format would make the test does not work properly., 

Recently TF-M introduced log levels and by default, the printing of the string is in interest was turned off. As a result FI test did not recognise the case when TF-M booted up properly, which means that FI protection code did not catch the injected fault. 

The following improvement is proposed:
- Decouple FI test from TF-M secure executable
- Build and use a fixed binary as a secure payload for the test. NS image does not need. 
- Fixed secure payload can be stored in the repo.
- Secure payload only prints a single line and jump to an infinite loop.
- FI test greps for this log message.
- FI test can be made more robust to check the successful booting at the beginning of the test without tampering with the image. PoC code is ready [here](https://github.com/matetothpal/mcuboot/commit/de4f5ed05d5140d6a28790337bb1f04dffca6966). If this fails, then the entire test fails.

tamasban, 2021-09-21:
@Sherryzhang2 Could you take a look at this?

Sherryzhang2, 2021-09-22:
So,  the secure image does not need to include TF-M, just build a secure project which can be a beta metal application based on an521 and then put the created image in this repo and modify the fault injection test script to grep the only log the secure printed. Is my understanding right?

tamasban, 2021-09-22:
Yes, you are right. However, there is another possibility to grep a message printed by MCUboot in a very late stage of the boot phase:
`[INF] Jumping to `
@matetothpal has done some PoC work, but we have noticed some strange results. Please contact him, he can share his code, but the results require some investigation.

matetothpal, 2021-09-22:
@Sherryzhang2 This is the commit @tamasban is referring to:
https://github.com/matetothpal/mcuboot/commit/6b2f281d582791d1648bc532f1280756df1e6e22

The strange thing with this is that there are some strange behaviours in the summary:
```
Tests failed to execute successfully (126):
    No instructions were skipped (e.g. branch was not executed), but booted successfully (126)
```
This means that there were some cases when no breakpoint was hit, i.e. no instructions were skipped, however the printout still appeared in the log, suggesting TF-M would have booted (we don't know whether it really could boot as the decision was made based on an MCUBoot printout). 

Not really sure how this could happen.

closed 2022-04-06 by github-actions[bot]