mcu-tools/mcuboot #1164: nRF52840DK - mcuboot does not load parent application

Choptec, 2021-10-04:
When compiling mcuboot tag: v1.7.99-ncs1 as a child image the application is not loaded. 

Board: nrf52840dk_nrf52840
Compiling: zephyr/samples/subsys/usb/cdc_acm/
Added "CONFIG_BOOTLOADER_MCUBOOT=y" > prj.conf
Added usb_cdc_acm_log_recovery.conf with the following contents: 
> CONFIG_LOG=y
> # The build won't fit on the partition allocated for it without size
> # optimizations.
> CONFIG_SIZE_OPTIMIZATIONS=y
> CONFIG_PM_PARTITION_SIZE_MCUBOOT=0x10000
> # Serial
> CONFIG_SERIAL=y
> CONFIG_UART_LINE_CTRL=y
> # MCUBoot serial
> CONFIG_MCUBOOT_SERIAL=y
> CONFIG_BOOT_SERIAL_CDC_ACM=y
> CONFIG_BOOT_MAX_LINE_INPUT_LEN=512
> CONFIG_BOOT_SERIAL_DETECT_PORT="GPIO_0"
> CONFIG_BOOT_SERIAL_DETECT_PIN=15
> CONFIG_BOOT_SERIAL_DETECT_PIN_VAL=0
> CONFIG_LOG_BACKEND_UART=y
> CONFIG_LOG_BACKEND_RTT=n
west build -b nrf52840dk_nrf52840 -p -- -Dmcuboot_OVERLAY_CONFIG=/home/user/ncs/zephyr/samples/subsys/usb/cdc_acm/usb_cdc_acm_log_recovery.conf

Result: 
When compiling the above sample with Nordic nRFconnect v1.7.0 the resulting mcuboot child image is unable to boot into the sample application. 
I tested further with the following nRFconnect versions: 
v1.7.0 - not working (mcuboot v1.7.99-ncs3)
v1.6.0 - not working (mcuboot v1.7.99-ncs2)
v1.6.0-rc1 - not working (mcuboot v1.7.0-ncs1-rc1)
v1.5.1 - works fine (mcuboot v1.7.99-ncs1)
v1.5.0 - works fine (mcuboot v1.7.99-ncs1)

When flashing the "known working" nRFconnect mcuboot from cdc_acm/build/mcuboot/zephyr/zephyr.hex over to the nrf52840dk device along with the cdc_acm/build/app_signed.hex from a "not working" build the application is able to boot. 
The only mcuboot version that has worked for me so far is v1.7.99-ncs1
As I missing some configuration for the child mcuboot image?

d3zd3z, 2021-10-04:
Just to clarify, are you saying the v1.7.99-ncs1 version does work, but 1.6.0-1.7.0 do not work?

Would it be possible for you to test the current release candidate v1.8.0-rc2? It'd be nice to know if we have something to fix for this release, or if this is just something that needs to be fixed on back releases.

It might also be useful to test #1124 to see if there is anything there that would fix this issue. It doesn't look like it, but it wouldn't hurt to try it.

Choptec, 2021-10-04:
Yes - v1.7.99-ncs1 does work when used with v1.5.1 of the nRFconnect SDK-bundled Zephyr version. 
With mcuboot v1.8.0-rc2 it doesn't compile mcuboot as a child image despite it being enabled in Kconfig. See below. 

> [samuel@ArchPC cdc_acm]$ west build -b nrf52840dk_nrf52840 -- -Dmcuboot_OVERLAY_CONFIG=/home/samuel/ncs/zephyr/samples/subsys/usb/cdc_acm/usb_cdc_acm_log_recovery.conf 
> -- west build: generating a build system
> Including boilerplate (Zephyr base (cached)): /home/samuel/ncs/zephyr/cmake/app/boilerplate.cmake
> CMake Deprecation Warning at /home/samuel/ncs/zephyr/cmake/app/boilerplate.cmake:37 (cmake_policy):
>   The OLD behavior for policy CMP0079 will be removed from a future version
>   of CMake.
> 
>   The cmake-policies(7) manual explains that the OLD behaviors of all
>   policies are deprecated and that a policy should be set to OLD only under
>   specific short-term circumstances.  Projects should be ported to the NEW
>   behavior and not rely on setting a policy to OLD.
> Call Stack (most recent call first):
>   /home/samuel/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
>   /home/samuel/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
>   CMakeLists.txt:4 (find_package)
> 
> 
> -- Application: /home/samuel/ncs/zephyr/samples/subsys/usb/cdc_acm
> -- Zephyr version: 2.6.99 (/home/samuel/ncs/zephyr), build: v2.6.99-ncs1
> -- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
> -- Board: nrf52840dk_nrf52840
> -- Cache files will be written to: /home/samuel/.cache/zephyr
> -- Found dtc: /usr/bin/dtc (found suitable version "1.6.1", minimum required is "1.4.6")
> -- Found toolchain: gnuarmemb (/home/samuel/gnuarmemb/gcc-arm-none-eabi-10-2020-q4-major)
> -- Found BOARD.dts: /home/samuel/ncs/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
> nrf52840dk_nrf52840.dts.pre.tmp:15.39-20.5: Warning (interrupt_provider): /soc/interrupt-controller@e000e100: Missing #address-cells in interrupt provider
>   also defined at nrf52840dk_nrf52840.dts.pre.tmp:370.7-372.3
> -- Generated zephyr.dts: /home/samuel/ncs/zephyr/samples/subsys/usb/cdc_acm/build/zephyr/zephyr.dts
> -- Generated devicetree_unfixed.h: /home/samuel/ncs/zephyr/samples/subsys/usb/cdc_acm/build/zephyr/include/generated/devicetree_unfixed.h
> -- Generated device_extern.h: /home/samuel/ncs/zephyr/samples/subsys/usb/cdc_acm/build/zephyr/include/generated/device_extern.h
> -- Including generated dts.cmake file: /home/samuel/ncs/zephyr/samples/subsys/usb/cdc_acm/build/zephyr/dts.cmake
> Parsing /home/samuel/ncs/zephyr/samples/subsys/usb/cdc_acm/Kconfig
> Loaded configuration '/home/samuel/ncs/zephyr/samples/subsys/usb/cdc_acm/build/zephyr/.config'
> Configuration saved to '/home/samuel/ncs/zephyr/samples/subsys/usb/cdc_acm/build/zephyr/.config'
> Kconfig header saved to '/home/samuel/ncs/zephyr/samples/subsys/usb/cdc_acm/build/zephyr/include/generated/autoconf.h'
> CMake Warning at /home/samuel/ncs/zephyr/subsys/usb/CMakeLists.txt:22 (message):
>   CONFIG_USB_DEVICE_VID has default value 0x2FE3.
> 
>   This value is only for testing and MUST be configured for USB products.
> 
> 
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/samuel/ncs/zephyr/samples/subsys/usb/cdc_acm/build
> -- west build: building application
> [161/168] Linking C executable zephyr/zephyr_prebuilt.elf
> 
> [168/168] Linking C executable zephyr/zephyr.elf
> Memory region         Used Size  Region Size  %age Used
>            FLASH:       49500 B       412 KB     11.73%
>             SRAM:       21088 B       256 KB      8.04%
>         IDT_LIST:          0 GB         2 KB      0.00%
>

closed 2022-04-17 by github-actions[bot]