mcu-tools/mcuboot #1165: Swap using scratch checks for swap size being less than primary/secondary slot sizes but this is incorrect

lairdjm, 2021-10-05:
At https://github.com/mcu-tools/mcuboot/blob/main/boot/bootutil/src/swap_scratch.c#L247 there is a check for if the primary slot or secondary slot sizes are greater than the scratch size and if so, it fails. The whole point in having the scratch area as far as I understand it is so that data from one slot can be put there whilst the other slot's data goes to the main slot and that the scratch area can be filled up so that one swap can contain many sectors of either area, speeding up the process of swapping as it doesn't do it on a sector by sector basis (after spending time shifting all sectors in one slot up by one)

We have the primary slot on main flash of an nRF52840 and secondary slot in QSPI - both have 4KB sector sizes, the scratch for testing is placed on QSPI also, current head of tree zephyr is being used at commit ef0d955758e549f39d97b07089dedc4bdd05bc0e - with this check in, upgrade using scratch is not possible (testing with 64KB slot sizes and 16KB scratch size), with that check commented out, the upgrade can proceed and works, image upgrading works, image reverting works, so it seems like a bug in that check

lairdjm, 2021-10-05:
Seems to be an issue with this however in that mcuboot will always swap the images at startup, even if you confirm the image from the application, on reboot mcuboot will swap them over.

de-nordic, 2021-10-05:
> At https://github.com/mcu-tools/mcuboot/blob/main/boot/bootutil/src/swap_scratch.c#L247 there is a check for if the primary slot or secondary slot sizes are greater than the scratch size and if so, [...]

The code you are pointing to is check whether scratch area is big enough to cover largest sector of primary/secondary slot; the check is done because flash, for example in stm32 devices, does not have to have sectors of the same size.
In case of uniform size sectors the `sz?` will always have size of sector here, and will be reset to 0 before leaving this if.

Something is wrong with your DTS configuration of slots. I have tested the feature with 4k (1 page/sector) size scratch on QSPI and secondary slot on QSPI and everything seems to work.
Are you sure that last sectors of your secondary slot are not overlapping with the scratch?

lairdjm, 2021-10-05:
The sector size of the nRF52840 and MX25R64 are both 4KB, the block size of the MX25R64 is 64KB but that size should not be being used. I've changed to using a 64KB scratch size and the behaviour of mcuboot/zephyr when mcuboot is built for scratch with swap operation is so bizarre, the DTS file is:
```
&flash0 {
        /*
         * For more information, see:
         * http://docs.zephyrproject.org/latest/guides/dts/index.html#flash-partitions
         */
        partitions {
                compatible = "fixed-partitions";
                #address-cells = <1>;
                #size-cells = <1>;

                /* 96K */
                boot_partition: partition@0 {
                        label = "mcuboot";
                        reg = <0x000000000 0x0000c000>;
                };
                slot0_partition: partition@c000 {
                        label = "image-0";
                        reg = <0x0000c000 0x000ec000>;
                };
                storage_partition: partition@f8000 {
                        label = "storage";
                        reg = <0x000f8000 0x00008000>;
                };
        };
};

&mx25r64 {
        partitions {
                compatible = "fixed-partitions";
                #address-cells = <1>;
                #size-cells = <1>;

                /* 896K */
                slot1_partition: partition@0 {
                        label = "image-1";
                        reg = <0x0000000 0x000ec000>;
                };
                /* 16K */
                scratch_partition: partition@Ec000 {
                        label = "image-scratch";
                        reg = <0x000EC000 0x00010000>;
                };
                lfs_partition: partition@100000 {
                        label = "lfs_storage";
                        reg = <0x00100000 0x000700000>;
                };
        };
};
```

I can load a signed firmware, I can load an app update using mcumgr, I can test that firmware using mcumgr but when I reboot, the status of the image that is being tested changes to confirmed and active and the other slot changes to pending, confirming the image does nothing, then when the module reboots, it swaps the images back and boots the other firmware, when it's finished and the old firmware boots, again the mcuboot configuration shows that the old version is confirmed and active and the other slot is pending - on every reboot it will just flip the image, for no reason that I can see. The only way I can get out of this mode is to erase the secondary slot prior to rebooting, this then causes mcuboot to change it's status to revert and then when that image boots, I can then confirm it and it will be the primary image, I can load another firmware update but if I do I go through the same problematic issues.

The swap using scratch upgrade method, despite seeming to be broken, is substantially faster than swap without scratch, we are seeing upgrade times of about 1 minute with mcuboot configured to use the CC310 and swapping without scratch of a 450KB image and upgrade times of about 25 seconds with mcuboot configured to use tinycrypt and swapping with the scratch partition with the same image

de-nordic, 2021-10-08:
Is the mcuboot build with exactly the same partition layout? Are you using the sdk-mcuboot?

> So why is the secondary slot still marked as pending if the image there cannot be used, the pending flag should be removed automatically by the bootloader surely because the image is incompatible?

Yes it should if image would fail checks.

nvlsianpu, 2021-12-21:
@lairdjm 
> So why is the secondary slot still marked as pending if the image there cannot be used, the pending flag should be removed automatically by the bootloader surely because the image is incompatible?

Pending image in the secondary slot can not be erased by mcumgr. Need to reboot the board so mcuboot will act so in boot_validate_slot() call invalid image will be erased. After that you should be able to run what you have in the primary slot.

d3zd3z, 2022-06-24:
> The code you are pointing to is check whether scratch area is big enough to cover largest sector of primary/secondary slot; the check is done because flash, for example in stm32 devices, does not have to have sectors of the same size.

Note that, as far as I can tell, MCUboot has never been used with differing sector sizes, at least in the same flash device. I think we've done external flash with a different sector size for primary and secondary. On the stm32, for example, we only use 3 128k sectors.

To clarify an earlier point, the scratch area is not an "optimization" to make it faster, it is needed in order to be able to recover, at any time, from a loss of power. Without at least one additional sector, there is no way to be able to do the swap and survive a power loss.

That said, _nobody_ should be using a configuration of mcuboot any more with a scratch area. Everyone should be using 'swap move', which does not use scratch, but merely requires that slot0 be exactly one slot larger than slot1. The largest image that can be upgraded will be the size of slot 1, or one sector less than the size of slot 0, whichever is smaller.

closed 2022-07-09 by github-actions[bot]