mcu-tools/mcuboot #1171: wrong free space check when doing swap?

novalisek, 2021-10-17:
Hi,
I am wokring with NRF52832. I have primary and secondary size partitions sizes set to 0x35000 (212kB)
My code size is 211604 (little under 208kB)
I am getting "Not enough free space to run swap upgrade". 
So I have put some debug info into swap_run function https://github.com/mcu-tools/mcuboot/blob/29099e1d17f93ae1d09fe945ad191b703aacd3d8/boot/bootutil/src/swap_move.c#L439 

so that part of it looks like this now:
```

 sz = 0;
    g_last_idx = 0;

    sector_sz = boot_img_sector_size(state, BOOT_PRIMARY_SLOT, 0);
	BOOT_LOG_WRN("%s - copy_size = %u",__PRETTY_FUNCTION__,copy_size);
	BOOT_LOG_WRN("%s - sector_sz = %u",__PRETTY_FUNCTION__,sector_sz);
    while (1) {
        sz += sector_sz;
        /* Skip to next sector because all sectors will be moved up. */
        g_last_idx++;
        if (sz >= copy_size) {
            break;
        }
    }
	BOOT_LOG_WRN("%s - sz = %u",__PRETTY_FUNCTION__,sz);
	BOOT_LOG_WRN("%s - g_last_idx = %u",__PRETTY_FUNCTION__,g_last_idx);

    /*
     * When starting a new swap upgrade, check that there is enough space.
     */
    if (boot_status_is_reset(bs)) {
        sz = 0;
        trailer_sz = boot_trailer_sz(BOOT_WRITE_SZ(state));
		BOOT_LOG_WRN("%s - trailer_sz = %u",__PRETTY_FUNCTION__,trailer_sz);
        first_trailer_idx = boot_img_num_sectors(state, BOOT_PRIMARY_SLOT) - 1;
		BOOT_LOG_WRN("%s - first_trailer_idx = %u",__PRETTY_FUNCTION__,first_trailer_idx);

        while (1) {
            sz += sector_sz;
            if  (sz >= trailer_sz) {
                break;
            }
            first_trailer_idx--;
        }
		BOOT_LOG_WRN("%s - first_trailer_idx = %u",__PRETTY_FUNCTION__,first_trailer_idx);

        if (g_last_idx >= first_trailer_idx) {
			BOOT_LOG_WRN("%s - g_last_idx: %u >= first_trailer_idx: %u",__PRETTY_FUNCTION__,g_last_idx,first_trailer_idx);
            BOOT_LOG_WRN("Not enough free space to run swap upgrade");
            bs->swap_type = BOOT_SWAP_TYPE_NONE;
            return;
        }
    }
```

And I am getting this output:

```
W: swap_run - copy_size = 211604
W: swap_run - sector_sz = 4096
W: swap_run - sz = 212992
W: swap_run - g_last_idx = 52
W: swap_run - trailer_sz = 1584
W: swap_run - first_trailer_idx = 52
W: swap_run - first_trailer_idx = 52
W: swap_run - g_last_idx: 52 >= first_trailer_idx: 52
W: Not enough free space to run swap upgrade
```

If I understand correctly, if my code is aligned to whole sectors, then size is 212992 (208kB), divided by 4kB sector size = 52 sectors totaly.  Partition size is 212kB (53 sectors). So it shoud fit (52 sectors code + 1 sector for trailer). Am I missing something? :)  

Should
```
if (g_last_idx >= first_trailer_idx) {
be just
if (g_last_idx > first_trailer_idx) {
```
? To me g_last_idx is not index, but number of sectors? (so index would be 1 less?)

Thanks.

nvlsianpu, 2021-12-21:
@novalisek

When using this algorithm the maximum image size available for the application will be:

`maximum-image-size = (N-1) * slot-sector-size - image-trailer-sectors-size`
Where:
`N` is the number of sectors in the primary slot.
`image-trailer-sectors-size` is the size of the image trailer rounded up to the total size of sectors its occupied. For instance if the image-trailer-size is equal to 1056 B and the sector size is equal to 1024 B, then `image-trailer-sectors-size` will be equal to 2048 B.

I added this note recently to the project documentation.
Looks like your exceeds the size limit calculates from above equation.

novalisek, 2021-12-21:
@nvlsianpu why is it N-1 sectors minus trailer size? you can use Nth sector for trailer, or is it used for something else? According to your formula, my image can only have N-2 sectors ... then one sector is used for trailer and the last one? 

[Documentation](https://docs.mcuboot.com/design.html) says  
maximum-image-size = image-slot-size - image-trailer-size

Thanks.

nvlsianpu, 2021-12-21:
You are using swap without swap algorithm. Don't you?

https://docs.mcuboot.com/design.html#image-swap-no-scratch

This algorithm is designed so that the higher sector of the primary slot is used only for allowing sectors to move up.
Also the image trailer needs its own sector. So N-2 sectors are available for the image.

novalisek, 2021-12-21:
@nvlsianpu that may be the problem - I didn't select any algorith in prj.conf, so may be defaults are used. And if I check https://github.com/mcu-tools/mcuboot/blob/main/boot/zephyr/Kconfig I see 

```
choice BOOT_IMAGE_UPGRADE_MODE
	prompt "Image upgrade modes"
	default BOOT_SWAP_USING_MOVE if SOC_FAMILY_NRF
	default BOOT_SWAP_USING_SCRATCH
```

so may be it is selecting SWAP_USING_MOVE if I use NRF52832 ?  I should specify CONFIG_BOOT_SWAP_USING_SCRATCH in prj.conf

nvlsianpu, 2021-12-21:
Yes, By default you have BOOT_SWAP_USING_MOVE selected.
You can use swap using scratch instead. I ques you want limit the scratch size to 1 sector size. Just be aware that the wear of the scratch flash will be increased much in compassion to the rest of the flash (This is elaborated in documentation fragment you pointed before).

novalisek, 2021-12-21:
ok, thanks. I will try it and close this issue if it helps

novalisek, 2021-12-22:
OK, so CONFIG_BOOT_SWAP_USING_SCRATCH=y  helped, so it was my fault I didn't notice the type of swap. May be mcuboot could log the type of swap while doing the swap (test swap or revert swap) at startup (or at least when it writes error that there is not enough space), so that you don't have to look into prj.conf or check KConfig what could the defaults be.

closed 2021-12-22 by novalisek

nvlsianpu, 2021-12-23:
> Possibly related to #1165.

Was not really related.

nvlsianpu, 2021-12-23:
@novalisek I added logs you had requested: https://github.com/mcu-tools/mcuboot/pull/1261