mcu-tools/mcuboot #1181: Just a question - Can ESP32 talk to MCUBoot ?

hassandraga, 2021-10-21:
We are planning to start using MCBoot soon, and I have some questions please.
Basically, our board has two MCU, ESP32 and STM32, we are planning to flash the STM32 with MCBoot, but ESP32 won't have the MCBoot. 

My question is, did can we make ESP32 securely talk to STM32(MCBoot)? or we needs to have MCUBoot on both sides?
`[ ESP32 ] ---> uart ---> [ STM32 (MCUBoot) ]`

The objective is to make ESP32 download the latest firmware for STM32, talk to MCUBoot and start handshake and flashing.

beriberikix, 2022-04-21:
Can someone remove the stale label? This is still a common request.

closed 2022-11-03 by github-actions[bot]

beriberikix, 2023-02-14:
While not exactly the original question, https://github.com/espressif/esp-serial-flasher/issues/34 might be relevant to some folks.