mcu-tools/mcuboot #1191: Size overflow for block size in image encryption

kvaellning, 2021-11-02:
**Issue:**

I am using MCUboot with Zephyr, and set a header size of 0x1000. In `loader:boot_copy_region`, this leads to a chunk size of 0x400 (buffer size). If on start of the image (`loader.c:907`), this leads to a silent overflow in `blk_sz` on line 910.

Interesting code:
`blk_sz = chunk_sz - hdr->ih_hdr_size; // chunk_sz = 0x400; hdr->ih_hdr_size = 0x1000`.

**Suggested fix:**

1. Check for this overflow, return `BOOT_EBADARGS`
2. imgtool should check for meaningful values of the header size.

nvlsianpu, 2021-11-02:
@kvaellning Is so big header something that you needed? Current code support header up to 1024 B size.
If this is just a bug hunting then I would go with fixes you suggested.

kvaellning, 2021-11-02:
This was basically a misconfiguration on my side, and I was curious why my image was corrupted without seeing an error. For my application, I can live with smaller header sizes (which I now use). I‘ll provide a pull request with the suggested fixes by tomorrow.

d3zd3z, 2021-11-03:
Adding a check is a good first step. The code could be reworked to not require that the header fit into the first block, though.

The header size is determined by the number of IRQs on the SoC (at least for Arm). The number of IRQs in the vector table determines the required alignment of the image (hence the size of the header). 256 IRQs isn't that uncommon, and I could see needing more.

nvlsianpu, 2021-11-04:
^^ I'm looking at better solution.

PR #1199 (2021-11-15): loader: Allow image header bigger than 1 KB for encrypted images
boot_copy_region() was written so it assumes that the image header
must fit int the intermediary buffer of 1 KB size. A bigger header
will cause a overflow in calculation of size of data chunk to be
decrypted.

This patch allow to use header bigger than that buffer size and
mitigate the limitation described above.

This patch was tested on zephyr nrf52840dk_nrf52840 with image headed size increased to 0x600.
It was done by simple modification i zephyr's Kconfig.zephyr file:

```Kconfig
config ROM_START_OFFSET
	hex
	prompt "ROM st [... cut]
closed 2021-11-15 by nvlsianpu