mcu-tools/mcuboot #1202: Image status is not set by flashing, only by doing DFU

apsigma, 2021-11-09:
I am working in a project where we use MCUboot with a Zephyr OS app on a nRF9160. The app supports FOTA/DFU via LwM2M, and utilizing the support in MCUboot with swapping primary/secondary.

I recently noticed a weird thing. It appears like my app image does not contain the image status.

If I erase my whole flash memory, and then flash my app, MCUboot reports the following status:
```
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none                                                                                                                                                                     
I: Swap type: none                                                                                                                                                                       
I: Bootloader chainload address offset: 0x10000
```

If I then do a FOTA/DFU, MCUboot reports when I reboot:
```
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=good, swap_type=0x2, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Swap type: test
I: Bootloader chainload address offset: 0x10000
```

If I then reboot or reflash, I get the following report from MCUboot:
```
I: Starting bootloader
I: Primary image: magic=good, swap_type=0x2, copy_done=0x1, image_ok=0x1
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Swap type: none                                                                                                                                                                       
I: Bootloader chainload address offset: 0x10000
```

Which I think is expected, since the DFU went well.

What I don't understand is why I get the `magic=unset` when I erase and then flash. Is that intended for usage of MCUboot? Or is that some problem in Zephyr/west?

Also, in our code, we check the image status, and then actually always forces it to be OK, since we don't do any testing currently. This works as expected when we do an DFU, but in the case where I have erased the flash memory, and then flashed, and the status is `magic=unset`, MCUboot doesn't allow setting the image status. In this snippet from [bootutil_public.c](https://github.com/nrfconnect/sdk-mcuboot/blob/main/boot/bootutil/src/bootutil_public.c), in function `boot_set_confirmed_multi`:

```c
    switch (state_primary_slot.magic) {
    case BOOT_MAGIC_GOOD:
        /* Confirm needed; proceed. */
        break;

    case BOOT_MAGIC_UNSET:
        /* Already confirmed. */
        goto done;

    case BOOT_MAGIC_BAD:
        /* Unexpected state. */
        rc = BOOT_EBADVECT;
        goto done;
    }
```

Why does it say `Already confirmed.` when it is `BOOT_MAGIC_UNSET`? What is the intended case where it is unset, and we try to update the image to be OK?

d3zd3z, 2021-11-16:
I believe this was intended so that a tool could later write the magic and cause MCUboot to revert the image. This would only work if rollback protection was not enabled.

apsigma, 2021-11-16:
Hmm, why would you want the possibility to revert the image in the case when you have done a manual flash (i.e. overwriting the primary slot), and not a DFU? And what would that revert to? If I erase the whole flash, the image in the secondary slot will not be valid.

d3zd3z, 2021-11-17:
I'm guessing it is an artifact of the "real swap" code and what DFU does. The DFU code was added after the swap code was written.

apsigma, 2021-11-18:
Oh, alright, that sounds reasonable.

closed 2021-11-18 by apsigma