mcu-tools/mcuboot #1203: Change Mbed TLS in sim to use Mbed TLS heap

d3zd3z, 2021-11-10:
Currently, the sim configures Mbed TLS to just use malloc/free. This misses failures that are caused by the Mbed TLS heap being too small. Change the configuration to match how this is built on targets, using the internal Mbed TLS heap with a fixed size. This should help catch issues like #1200 earlier.

utzig, 2021-11-10:
Given that the simulator is built for x86_64 and the usual target for Zephyr is Cortex-Mx, I am not sure it will have use the exact same numbers for heap; also this feels a bit like testing Zephyr and not the bootloader itself...

d3zd3z, 2021-11-10:
I agree it isn't exactly the same, and I just thought of another reason it is kind of challenging: the sim is multi-threaded. This works ok with the larger system heap, but doesn't really make sense to use a smaller heap when multiple threads are using it.

I don't think it is really a zephyr-specific thing, although the configs are Zephyr specific (I'm not quite sure how the configuration is evening happening with the other platforms).

I'll put this as a someday thing, since it is kind of quite a bit of work for something that would be revealed pretty quickly on target.