mcu-tools/mcuboot #1209: Reference to End User License Agreements - EULA not found

rahulmohang, 2021-11-13:
Hi, 

There are references to End User License Agreements in some of the files copyrighted by Cypress Semiconductor Corporation. However, the EULA has not been found in the source package. Path to those files:

mcuboot/boot/cypress/MCUBootApp/os/os.h
main/boot/cypress/cy_flash_pal/cy_smif_psoc6.c
main/boot/cypress/cy_flash_pal/include/cy_smif_psoc6.h
main/boot/cypress/cy_flash_pal/flash_qspi/flash_qspi.c
main/boot/cypress/cy_flash_pal/flash_qspi/flash_qspi.h

Can you please let me know about the EULA that is being referred? The project that I represent uses the files, and since the project is an open source project, we are looking forward to avoid using source files licensed under licenses that are not open source.

d3zd3z, 2021-11-17:
@romanjoe Can you look into what is needed here? At a minimum, we should include the EULA. It would be better if you could get Infineon to release these files under the Apache license, to match with the rest of MCUboot.

rahulmohang, 2021-12-06:
Hello Roman @romanjoe, 
Can you please let me know if you could find the EULA?

rahulmohang, 2022-06-22:
@d3zd3z , is it possible to reopen the issue?

rahulmohang, 2022-06-22:
Thank you @nvlsianpu.

d3zd3z, 2022-09-22:
This really needs to be handled by the owners of the files. There isn't much I can do, since I don't have the ability to change the licenses on files by other authors.