mcu-tools/mcuboot #1211: Software based downgrade prevention with single application slot

mateo-ligne-netatmo, 2021-11-15:
I am working on a project using MCUBoot (v1.7.99-ncs3) where my microcontroller will receive updates through UART from another chip using mcumgr library. 

I would like to use both following configuration: 
    • CONFIG_SINGLE_APPLICATION_SLOT
    • CONFIG_MCUBOOT_DOWNGRADE_PREVENTION

Currently the downgrade prevention option compare the version of the two stored firmware, but in my case I would like to compare the version of the new firmware with stored one before overwriting it.
   
Has this use case been considered? Is it possible to read the firmware header before loading it? 

Thanks for your answers, 
Matéo

d3zd3z, 2021-11-17:
When MCUboot is configured for single application slot, the downgrade prevention code wouldn't ever be run, as it is part of the upgrade process, which MCUboot isn't really involved in. This check you are requesting would need to be added to the serial recovery code.

However, the request is reasonable, and I don't see any reason this couldn't be implemented there.

mateo-ligne-netatmo, 2021-12-08:
Hi, 
I'm implementing the comparison of software version between stored firmware and firmware to be send over recovery UART from MCUmgr. Currently, I am able to allow / disallow flash write considering version but the issue I am facing is that in case of a blocked update, MCUmgr keep trying to send the firmware unless I fake the reception by incrementing offset. 

Do you have any return code or other solution that should stop an ongoing MCUmgr update ? 

Thanks for your answers,
Matéo

d3zd3z, 2021-12-13:
It may have just been an unexpected situation from the perspective of MCUmgr. Perhaps we do need to define an additional error code to indicate that the image has been rejected (I could see images being rejected for other reasons, say it is able to tell it is for the wrong device). MCUmgr (and probably newtmgr) might have to be modified to support this, though.

mateo-ligne-netatmo, 2021-12-16:
In linked PR #1239, I added this new error code that I called MGMT_ERR_REJECTED_UPD, it might be renamed and defined in several other files as I only used it in a Serial recovery configuration. Otherwise I made a development on newtmgr (https://github.com/apache/mynewt-newtmgr/pull/187) stopping the firmware sending on the reception of this newly defined code.

closed 2022-06-29 by github-actions[bot]