mcu-tools/mcuboot #1216: porting API for architecture-native image startup address check

nvlsianpu, 2021-11-19:
https://github.com/mcu-tools/mcuboot/pull/1206 introduces check which is working for ARM cores

Rationale:
isolate architecture specific code from mcuboot's core code, of which kind loader.c should be.

Objective:
Introduce interface API which will be for check whether application startup address belongs to the primary slot flash area.

Refine check implementation for ARM. Empty implementation for others cores.

d3zd3z, 2021-11-19:
I think there are multiple opportunities to do this within the code (not that we shouldn't do this one), our code has become fairly ifdef heavy, and I think there are enough of them sprinkled throughout the code that it seems like the obvious way to add a new feature. Ones to start with are ones like this, where the ifdefs are in the middle of blocks of code (the multi image stuff has a lot of these).

d3zd3z, 2022-05-19:
Removing the stale, as I think we do want to do this.

closed 2022-12-01 by github-actions[bot]