mcu-tools/mcuboot #1225: Doc | Bad trailer description

Aukstkalnis, 2021-11-30:
Hi. I was analyzing image's trailer functionality and noticed that it do not match with
with code. When I see this structure:
[image]
Swap size, is 4B but in code 8B,
Swap info is 7B but in code 8B,
Copy done is 7B but in code 8B,
Image OK is 7B but in code 8B.

Why is that? Or do I not understand this picture correctly? :-)

Bonus qestion.

What will hapen if image's size is in trailers sector? Will it be OK, or image can not enter trailers sector?

nvlsianpu, 2021-11-30:
You interpreted table incorectly, for instance:
image-ok is 1 B and remaining 7 B are the padding for ensure alignment.

Aukstkalnis, 2021-11-30:
Ok, and what about swap size? It is 8B long and not 4 as shown in the table.

utzig, 2021-11-30:
> Ok, and what about swap size? It is 8B long and not 4 as shown in the table.

Commit https://github.com/mcu-tools/mcuboot/pull/1217/commits/17042588880ab651e407fbc4bc77e61a94a469d8 (part of #1217) will fix these documentation issues, just read it there.

Aukstkalnis, 2021-11-30:
But I still dont undestand if i need to leave all sector for trailing data or image data can be writen up to trailer's start offset...

utzig, 2021-11-30:
If you're using `imgtool` it will correctly calculate how big an image can be (for swap-with-scratch), are you using something else to generate images?

Aukstkalnis, 2021-11-30:
No, I am using imgtool. I am asking this because I am bilding shell command that shows how many space is left and I need to know how log is a trailer.

utzig, 2021-11-30:
Then you'll have to call `boot_trailer_sz` and subtract that from the slot size, don't do the calculation yourself...

Aukstkalnis, 2021-11-30:
@utsig thanks for the tips. But I am using oder mcuboot version that do not have this routine. But it was goot tip to have a look at it :) I am closing this issue, because all questions were answered.