mcu-tools/mcuboot #1243: secp256k1 ECDSA signing/verification and key generation in mcuboot

Harshal248, 2021-12-15:
How could I sign the image with ECDSA signature key based on secp256k1 curve ?

utzig, 2021-12-15:
This is not supported by MCUboot as of yet but it should be straight-forward to do.

* The `cryptography` library which `imgtool` uses supports the curve https://cryptography.io/en/latest/hazmat/primitives/asymmetric/ec/#cryptography.hazmat.primitives.asymmetric.ec.SECP256K1 so you can just add a new class for it based on `ECDSA256P1` and build a secp256k1 curve by changing this line https://github.com/mcu-tools/mcuboot/blob/main/scripts/imgtool/keys/ecdsa.py#L93

* `tinycrypt` does not support it, but it's a matter of adding the proper parameters since this curve has the same form that secp256r1. Also as far as I remember `micro-ecc` supports it and this was the elliptic curve library that was used to create tinycrypt.

* Mbed TLS already has the support so it's just a matter of updating the configuration to enable it.

Unfortunately `ring` which is used by the simulator doesn't support this curve afaik and you would have to add it if you plan on integrating the upstream support, but if you're just planning on maintaining your changes then hopefully the pointers above should get you started.

Harshal248, 2021-12-15:
I have generated secp256k1 ECDSA key using openssl and used it into mcuboot conf. Then it gives me error for unsupported EC curve for secp256k1 in imgtool. 
For that I have made changes into imgtool to enable secp256k1 curve. Therefore, It can build that project. Then I have enabled MBEDTLS config for secp256k1 curve(CONFIG_MBEDTLS_ECP_DP_SECP256K1_ENABLED) and dependencies. After building the project, on flashing it stuck into bootloader by stating : "Image in the primary slot is not valid!" and stops booting.

Any resolution for this Invalid sign key error ?

utzig, 2021-12-16:
Oh, my comment above was just in regards to pre-requisites. You still have to add support in the verification code, add the TLV for the signature type to the image, etc. If you just want to hack something you could modify image_ec256.c to use the secp256k1 oid instead, and do the same for imgtool, and so on. Not clean but for testing it might suffice.

Do you have a requirement on using `secp256k1`? `secp256r1` should provide the same security properties.

d3zd3z, 2021-12-16:
One thing important, if you'd like to use 256k1, would be to find an implementation of it. The primary benefit of 256k1 over 256r1 is that the implementation can be made more efficiently. This is done at a potential cost of the security of the curve. If we don't have an implementation tuned to the curve, there isn't much point in using it.

So, I would first make sure you can find an implementation that is both small (similar to tinycrypt/uECC) and more efficient than what we've got. Otherwise, I'd say there isn't really any point in supporting the additional curve.

closed 2022-06-29 by github-actions[bot]