mcu-tools/mcuboot #1257: Support X.509 certificates for verification

d3zd3z, 2021-12-21:
Allow MCUboot to support X.509 certificate chains instead of just single public/private key pairs. This would allow one or more root certs to be stored in MCUboot, and more flexibility for those that are signing firmware images.

prayassamriya, 2022-03-18:
@d3zd3z can you provide some insights if the work is in progress here? any timelines to get this feature released?

d3zd3z, 2022-04-06:
I have a partial implementation based on the state of the tree some months ago. It isn't currently a high priority for me, and the current plans are to reconsider this after incorporating support for SUIT.

prayassamriya, 2022-04-07:
Hi @d3zd3z,
Thanks for the response. I would like to collaborate if you can share the high level design. we are thinking of some custom solution as this feature is not yet available. if in some way, we can contribute into the same design which will be part of mcuboot in the end, it will a win-win for both. do share your thoughts,

d3zd3z, 2022-04-18:
Feel free to look at #1351, I've pushed where things were last year. The code has changed a good bit, and it is missing code to actually compare the signature with the key in the certificate.

prayassamriya, 2022-04-19:
Thanks @d3zd3z.

VineetaNarkhede-eaton, 2022-05-13:
Hi @d3zd3z 

This is in regards to above comments by @prayassamriya .
I looked into https://github.com/mcu-tools/mcuboot/pull/1351
I believe that the certificate chain and image verification workflow would be like: 
[image]

As per my observation, pull request has a code to verify Intermediate certificate using Root key. Product key verification and image signature verification (yellow highlighted workflow in above flowchart) is missing, are you talking about the same missing code in your comment above?

To store certificate in a chain after parsing, shall we parse certificates in below order (should be managed by TLV sequence)
1. Root cert 
2. Intermediate cert 
3. Product cert
In pull request https://github.com/mcu-tools/mcuboot/pull/1351, Intermediate cert is added first then Root cert is added to the chain at the end.

I tried to integrate this PR changes in MCUboot code (version 1.8.0) and found that,
1. mbedtls is not allowed to enable for BOOT_SIGNATURE_TYPE_ECDSA_P256, hence kconfig file need to be modified.
2. CONFIG_BOOT_X509 symbol definition in kconfig need to be added.

Few more queries below:
1. is below TLV sequence correct? what would be your recommendation?
    IMAGE_TLV_X509 (Intermediate certificate)
    IMAGE_TLV_X509 (Product certificate)
    IMAGE_TLV_SHA256 (image hash)
    IMAGE_TLV_ECDSA256 (image signature)

2. What is expected to be done in **verify callback**
3. Root certificate is found in two files i.e.  _boot/zephyr/root_cert.c_ and _sim/mcuboot-sys/csupport/keys.c_
    bootutil_root_cert[] array in root_cert.c is accessed during validation, what is the use of root_x509_der[] array in keys.c
4. We want to store only Root key in device (embed in bootloader). Any reason for storing complete Root certificate in device?
5. If you have any flowchart or documentation regarding this x509 cert chain validation, could you please share that.

prayassamriya, 2022-05-24:
@d3zd3z Can you please share your thoughts on comments from @VineetaNarkhede-eaton. We would like to implement this functionality and waiting for your response. your guidance will help us implement this feature and will be part of mcuboot.

vipulkute-eaton, 2022-07-20:
Hi @d3zd3z,

We wanted to understand how in certificate based approach we are planning to generate signed image.
As we wanted to contribute on the code signing functionality,  we did some analysis on the script side and come up with one easy solution to add certificate TLV in signed image. 
[image]

To achieve this we added one command "--cert" where we are passing the tag name and certificate file path. Since we need 3 Certificate ( Root, Intermediate and Product ) we need to write "--cert" three times followed by tag name and file path.
Tag field is added in command to provide any specific tag name to the TLV if required.
Ex:  
--cert Tag_field root_certificate_path --cert Tag_Field intermediate_certificate_path --cert Tag_Field product_certificate_path

Please let us know your input on the points mention above asap.

closed 2023-01-31 by github-actions[bot]