mcu-tools/mcuboot #1262: espressif: Bootloader size limitation when Flash Encryption + Secure Boot V2 + MCUboot signing are enabled

almir-okato, 2021-12-28:
This issue was created to track the following scenario:

Both Flash Encryption and Secure Boot V2 hardware-powered support were added to the Espressif port in addition to the MCUboot security chain.
Each feature alone can be enabled without issues. However, currently when all features are enabled at the same time, the resulting bootloader binary size for some chip + feature combination may exceed the size limit for the ROM bootloader verification, which may prevent the system from boot ([Documentation](https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32/security/secure-boot-bootloader-size.html#secure-boot-bootloader-size)).

Considering the signing schemes currently implemented by MCUboot, these are the combinations currently supported by the Espressif port:  

<table>
<thead>
  <tr>
    <th rowspan="2">MCUboot signing scheme</th>
    <th colspan="3">No security features</th>
    <th colspan="3">Secure Boot V2 + Flash Encryption</th>
  </tr>
  <tr>
    <th>ESP32</th>
    <th>ESP32-S2</th>
    <th>ESP32-C3</th>
    <th>ESP32</th>
    <th>ESP32-S2</th>
    <th>ESP32-C3</th>
  </tr>
</thead>
<tbody>
  <tr>
    <td>RSA-2048</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
    <td> :x: </td>
  </tr>
  <tr>
    <td>RSA-3072</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
    <td> :x: </td>
  </tr>
  <tr>
    <td>EC256</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
  </tr>
  <tr>
    <td>ED25519</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
    <td>:white_check_mark:</td>
    <td> :x: </td>
    <td> :x: </td>
    <td>:white_check_mark:</td>
  </tr>
</tbody>
</table>

d3zd3z, 2022-01-11:
At least some of the Nordic devices also have restrictions on the size of the bootloader.

closed 2022-08-11 by github-actions[bot]

zhang-wenchao, 2025-02-27:
I just tested esp32c3 ed25519 and it doesn't work either

[esp32c3] [ERR] [esp_image] Image length 69632 doesn't fit in partition length 65536
[esp32c3] [ERR] [secure_boot_v2] bootloader image appears invalid! error 8194