mcu-tools/mcuboot #1273: Need help for image encryption using STM32F7

krisnaresi, 2022-01-11:
Hello,

I've been stuck to add encryption feature for about several days, and I need some kind of help or advice.

So here is my design:

I'm using STM32F769. The memory region is looks like this:

```
Bootloader	Start: 0x8000000 Size: 0x8000
Primary		Start: 0x8040000 Size: 0x80000
Secondary	Start: 0x80C0000 Size: 0x80000
Swap		Start: 0x81C0000 Size: 0x40000
```

Several macro that I've defined:

```
#define MCUBOOT_SIGN_EC256
#define MCUBOOT_ENCRYPT_EC256
#define MCUBOOT_ENC_IMAGES
#define MCUBOOT_SWAP_USING_SCRATCH 1
#define MCUBOOT_VALIDATE_PRIMARY_SLOT
#define MCUBOOT_USE_TINYCRYPT 1
#define MCUBOOT_USE_FLASH_AREA_GET_SECTORS 1
#define MCUBOOT_HAVE_LOGGING
```

Key that I've defined:

```
/* For signing */
extern const unsigned char	ecdsa256_der[];
extern unsigned int			ecdsa256_der_len;
# define KEY_BUFFER			ecdsa256_der
# define KEY_LEN			ecdsa256_der_len

/* For encrypting */
extern const unsigned char	ecdsa256_enc_der[];
extern unsigned int			ecdsa256_enc_der_len;
# define KEY_BUFFER_ENC		ecdsa256_enc_der
# define KEY_LEN_ENC		ecdsa256_enc_der_len

const struct bootutil_key bootutil_keys[] = {
    [0] = {
        .key = KEY_BUFFER,
        .len = &KEY_LEN,
    }
};
const int bootutil_key_cnt = sizeof(bootutil_keys)/sizeof(bootutil_keys[0]);

const struct bootutil_key bootutil_enc_key = {
    .key = KEY_BUFFER_ENC,
    .len = &KEY_LEN_ENC,
};
```

As you can see I'm going to use ECDSA256 key type for both signing and encrypting, but the key value is different. Here is the imgtool script that I used to generate both sign and sign+encrypt image.

```
/* Sign only image */
python imgtool.py sign --key /src/lib/platform/boot/keys/ecdsa256.pem --header-size 0x200 --align 4 --slot-size 0x80000 --version 0.0.0 --pad-header --pad application.bin application-signed.bin
```

```
/* Sign and encrypt image */
python imgtool.py sign --key /src/lib/platform/boot/keys/ecdsa256.pem --header-size 0x200 --align 4 --slot-size 0x80000 --version 0.0.0 --pad-header --pad --encrypt /src/lib/platform/boot/keys/ecdsa256_enc.pem application.bin application-signed-encrypted.bin
```

I update the secondary partition using Ymodem, upon success, I trigger the update using `boot_set_pending(1)` and call `HAL_NVIC_SystemReset()` to restart the system.

Here is when the weird thing happen.

When I update using signed only image, it works perfectly. Here is the log:

```
[DBG] flash_area_open: ID=1
[DBG] flash_area_open: ID=2
[DBG] flash_area_open: ID=3
[DBG] flash_area_open: ID=1
[DBG] flash_area_open: ID=2
[DBG] flash_area_open: ID=1
[DBG] flash_area_open: ID=3
[INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[INF] Boot source: primary slot
[DBG] flash_area_open: ID=1
[DBG] flash_area_open: ID=1
[DBG] flash_area_open: ID=2
[INF] Swap type: perm
[DBG] flash_area_open: ID=2
[DBG] flash_area_open: ID=2
[DBG] flash_area_open: ID=1
[DBG] flash_area_open: ID=2
[DBG] flash_area_open: ID=1
[DBG] flash_area_open: ID=2
[DBG] flash_area_open: ID=3
[DBG] erasing scratch area
[DBG] flash_area_erase: Addr: 0x081c0000 Length: 262144
[DBG] initializing status; fa_id=3
[DBG] flash_area_open: ID=2
[DBG] writing swap_info; fa_id=3 off=0x3ffd8 (0x81fffd8), swap_type=0x3 image_num=0x0
[DBG] flash_area_write: Addr: 0x081fffd8 Length: 4
etc, etc, etc,
```

But when I update using signed+encrypt image:

```
[DBG] flash_area_open: ID=1
[DBG] flash_area_open: ID=2
[DBG] flash_area_open: ID=3
[DBG] flash_area_open: ID=1
[DBG] flash_area_open: ID=2
[DBG] flash_area_open: ID=1
[DBG] flash_area_open: ID=3
[INF] Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[INF] Boot source: primary slot
[DBG] flash_area_open: ID=1
[DBG] flash_area_open: ID=1
[DBG] flash_area_open: ID=2
[INF] Swap type: perm
[DBG] flash_area_open: ID=2
[DBG] flash_area_open: ID=2
[DBG] flash_area_erase: Addr: 0x080c0000 Length: 524288
[ERR] Image in the secondary slot is not valid!
[DBG] flash_area_open: ID=1
[DBG] flash_area_open: ID=1
```

Why the bootloader erase address `0x080c0000` where secondary slot is reside? Of course the image is not valid as it has been deleted.

What I've been tried:

- Use `#define MCUBOOT_USE_MBED_TLS` instead `#define MCUBOOT_USE_TINYCRYPT 1` and add several source file. No luck, still the same log.
- Use ST-Link Utility and write secondary slot directly without Ymodem. Still the same log, but instead `[INF] Swap type: perm`, it shows `[INF] Swap type: test`.

Any suggestion is pretty much appreciated. Thank you for your time and assistance.

d3zd3z, 2022-01-11:
Unfortunately, these kinds of issues can be challenging to debug, especially since it likes to erase the image when it finds it is invalid. A good starting point would be to comment out the `flash_area_erase` calls in loader.c (753, 770, and 811 in tip code). Maybe even add a log message to these to see why it is deciding the upgrade is invalid.

Do you have a debugger available for your target? Otherwise, print messages may be necessary to learn more. At least by commenting out the erase calls, you can avoid having to reprogram the image every time.

krisnaresi, 2022-01-12:
Thank you for your reply.

From your comment, I traceback and found out the one that decide the upgrade is invalid is in `parse_ec256_enckey` from `encrypted.c` line 739. From the comment line, I see that I need to feed private key. The problem is, I feed the public key instead private key.

I do read the docs [encrypted_images.md(creating-your-keys-with-imgtool)](https://github.com/mcu-tools/mcuboot/blob/main/docs/encrypted_images.md#creating-your-keys-with-imgtool), it's told how to extract public key using `getpub` but no explanation about how to use `getpriv`, which is very weird because we don't use public key on image encryption. It's make newbie like me trapped using key generated from `getpub`. So my recommendation is to change the documentation to use `getpriv` instead `getpub`.

Before I close this issue, out of curiosity, why you guys use signing key algorithm that have asymmetric keys for image encryption that usually using symmetric key? Not to mention that you need to use the same key type for signing and encryption, which make it more weird.

utzig, 2022-01-12:
> I do read the docs encrypted_images.md(creating-your-keys-with-imgtool), it's told how to extract public key using getpub but no explanation about how to use getpriv, which is very weird because we [...]

Feel free to submit a PR fixing it.

> Not to mention that you need to use the same key type for signing and encryption, which make it more weird.

No, you have to create a different keypair for each, one for signing and another for encryption. Reusing the same keypair basically breaks all security, and in that case you might as well not use any security features.

closed 2022-01-14 by krisnaresi

xianghui-renesas, 2022-09-03:
Hi, does anyone know if this is fixed in MCUBoot 1.9? How can we generate custom encryption key array enc_key[] now? thanks!