mcu-tools/mcuboot #1275: boot/zephyr fails testing on min-write-block-size = <16>

paulwuertz, 2022-01-11:
### Description:

I flashed mcu_boot and the smp_srv sample from the zepyhr project with udp transport.
I checked that smp_srv's img_mgmt routines correctly wrote to the second slot of image one.
But still, when marking the second slot for testing on a reboot the swap ends in an `panic!` log, followed by a freeze.
A reboot tries to resume the img swap and ends in a bus fault.
After this the trailer of slot 2 needs to be erased/reset to boot into a normal image again.

### Findings:

I debugged both the `panic!`-log branch and the bus-fault case.
The actual swap is performed without any errors as I could observe. 
When writing the copy_done flag the function `swap_set_copy_done` in `loader.c` returns with an error code, since 
```
    if (align > BOOT_MAX_ALIGN) {
        return -1;
    }
```
in `bootutil_public.c:348` returns -1. This way the bootloader never boots an image again, since it cannot mark the swap as done.

utzig, 2022-01-11:
I don't think the public routines are completely updated, still needs to be done.

paulwuertz, 2022-01-11:
I am not sure, if there was an option to configure the write-block-size to 16.
I read through the last commits, issues and PRs and it seems I eyes some suggestions that a config would be needed.
Dont know if this already exsist, I did not find it yet in the docu.

The support for flash with write-block-size = 16 and 32 is quite recently, did some test the zephyr port already with a min-write-block-size>8 board?

I was also unsure if this is potentially an issue in zephyr, but I suppose it's from this repo, since I flashed the zephyr port of mcuboot from this repo, right?

paulwuertz, 2022-01-11:
Can one support you updating the public routines - I do not know if PRs for that are welcome or already on someones desk/TODO list?
I do not quite understand yet which "public routines" all need to be updated, but if I can help updating and have a working bootloader faster I would like to contribute if contributions are welcome ^^

utzig, 2022-01-11:
> I would like to contribute if contributions are welcome

I was going to do it, but you're welcome to take it over because my free time is almost none.

paulwuertz, 2022-01-11:
Is there some issue or rough guide on which functions need an update?
Otherwise I would search for `BOOT_MAX_ALIGN` and see if all uses hold for the use with min-write-size of 16 and 32...

paulwuertz, 2022-01-11:
And thank you for maintaining mcuboot, really appreciate your work!

I hope I can contribute rather then make more trouble asking questions :s

mnkp, 2022-01-28:
> I don't think the public routines are completely updated, still needs to be done.

I believe the support on MCUboot side is fairly complete. What is missing is a bit of glue code to allow setting of `MCUBOOT_BOOT_MAX_ALIGN` from Zephyr. For now this needs to be done by hand, e.g. by adding
```
-DCMAKE_C_FLAGS="-DMCUBOOT_BOOT_MAX_ALIGN=16"
```
to the build command. That should be safe, since `CMAKE_C_FLAGS` as a command line option to cmake should append and not overwrite C compiler flags.

Also `include/dfu/mcuboot.h` in Zephyr hardcodes `BOOT_MAX_ALIGN` as 8, needs to be fixed.

mnkp, 2022-01-28:
@utzig As far as I can tell the only missing code on MCUboot side to support flash with large write alignment on Zephyr is something like
```
#ifdef CONFIG_MCUBOOT_BOOT_MAX_ALIGN
#define MCUBOOT_BOOT_MAX_ALIGN CONFIG_MCUBOOT_BOOT_MAX_ALIGN
#endif
```
added to `boot/zephyr/include/mcuboot_config/mcuboot_config.h`.

Then there is some more work on the Zephyr side. We would need to add `MCUBOOT_BOOT_MAX_ALIGN` Kconfig option and clean up `include/dfu/mcuboot.h` in Zephyr.

In the project I'm working on we are using MCUboot with Atmel SAM device. It has flash with alignment 16. I am happy to spend some time to add missing bits and pieces so we can use an official version rather than our patched one. That is, I can prepare a PR, however I'm not sure how the work should be synchronized with the development on the Zephyr side.

paulwuertz, 2022-01-31:
I managed to make McuBoot run on the Atmel SAM by changing BOOT_MAX_ALIGN, as mentioned in [other issues] finally :) (https://github.com/mcu-tools/mcuboot/issues/713).

What confused me, was that I already tried to change that value before, but in the mcuboot repo I cloned,
not in the bootloader file included in the `~/zephyrproject/bootloader`. I cloned mcuboot in some project directory and changed the BOOT_MAX_ALIGN in the mcuboot repos bootutil_public.h. 

I did not notice that the my cmake found the unmodified zepyhr headers in my home directory first. But changing BOOT_MAX_ALIGN was all that was necessary in the end...

```diff
diff --git a/boot/bootutil/include/bootutil/bootutil_public.h b/boot/bootutil/include/bootutil/bootutil_public.h
index ab4f9c5..1d2505c 100644
--- a/boot/bootutil/include/bootutil/bootutil_public.h
+++ b/boot/bootutil/include/bootutil/bootutil_public.h
@@ -70,7 +70,7 @@ extern "C" {
 /** Swapping encountered an unrecoverable error */
 #define BOOT_SWAP_TYPE_PANIC    0xff
 
-#define BOOT_MAX_ALIGN          8
+#define BOOT_MAX_ALIGN          16
 
 #define BOOT_MAGIC_GOOD     1

```

closed 2022-08-15 by github-actions[bot]