mcu-tools/mcuboot #1285: Problem with checking ESP_LOAD_HEADER_MAGIC for ESP32.

boris92git, 2022-01-21:
I tried to build MCUBoot for ESP 32 how it writes in the manual(https://github.com/mcu-tools/mcuboot/blob/main/docs/readme-espressif.md)
And I got following error:
[esp32] [ERR] Load header magic verification failed. Aborting
I looked up on code and finded strange things.
imgtool inserts a number different from the one MCUBoot expects for esp32.
https://github.com/mcu-tools/mcuboot/blob/7679215d00a727a4065f81b50efe845274f7692f/boot/espressif/hal/include/esp_mcuboot_image.h#L12
https://github.com/mcu-tools/mcuboot/blob/7679215d00a727a4065f81b50efe845274f7692f/scripts/imgtool/image.py#L41
How is this supposed to work? Is a special version of imgtool used for ESP32?

utzig, 2022-01-25:
> https://github.com/mcu-tools/mcuboot/blob/7679215d00a727a4065f81b50efe845274f7692f/boot/espressif/hal/include/esp_mcuboot_image.h#L12 [...]

https://github.com/mcu-tools/mcuboot/blob/main/boot/bootutil/include/bootutil/image.h#L41

This is the MAGIC for MCUboot images, which is what `imgtool` adds because it creates images compatible with MCUboot. The `ESP_LOAD_HEADER_MAGIC` looks like a previous step to load MCUboot itself, I assume it's something like an internal bootloader who chainloads MCUboot as a second stage bootloader.

/cc @almir-okato @gustavonihei

almir-okato, 2022-01-25:
@boris92git, the `ESP_LOAD_HEADER_MAGIC` is prepend to the application image binary and is used to identify a header containing the load information for the "Espressif port compatible images" only.

If you are using NuttX or Zephyr RTOSes, you must ensure that you enabled the MCUboot build option for them, so this information will be placed in the resulting binary. After that you must sign the binary using `imgtool`, that will add the MCUboot header.
The encapsulation will be like this (external --> internal):
( MCUboot header ( ESP load header ( image binary

First the MCUboot header is parsed and the image is validated by the MCUboot process, then the ESP load header is parsed in order to load the image correctly, finally the image is loaded and started.

Which RTOS are you using?

boris92git, 2022-01-26:
> @boris92git, the `ESP_LOAD_HEADER_MAGIC` is prepend to the application image binary and is used to identify a header containing the load information for the "Espressif port compatible images" only. [...]

My project is based on esp-idf.

rftafas, 2022-01-26:
Short answer: 

ESP-IDF does not support MCUboot and MCUboot does not support FreeRTOS.

---

Long answer:

Espressif port for MCUboot is a bare metal bootloader. As IDFboot is also a baremetal bootloader, in theory it is possible to replace IDFboot with MCUboot. One should expect a fair amount of changes on ESP-IDF buildsystem. This seems the closest path to achieve that.

On the other hand, it is possible to add support for FreeRTOS on MCUboot (the OS Bootloader type) and then commit that work to both MCUboot and FreeRTOS main repos. Once that is done, considering that ESP-IDF contains a fork of FreeRTOS, it is also possible to have this changes ported to such fork. In this case, it would still require a baremetal bootloader (in this case, should keep IDFboot) and still require changes on ESP-IDF buildsystem.

rftafas, 2022-01-26:
PTAL: https://github.com/FreeRTOS/Lab-Project-FreeRTOS-MCUBoot

boris92git, 2022-01-28:
> 0xace637d3

Thanks a lot!

closed 2022-01-28 by boris92git