mcu-tools/mcuboot #1297: MCUboot v1.9.0-rc1 fails to build with Zephyr

mnkp, 2022-01-27:
Building MCUboot v1.9.0-rc1 with Zephyr fails early in the build process
```
mcuboot/boot/zephyr/build$ git describe  --dirty
v1.9.0-rc1
mcuboot/boot/zephyr/build$ cmake -GNinja -DBOARD=efr32_radio_brd4104a ..
Including boilerplate (Zephyr workspace): ~/work/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: ~/work/zephyrproject/mcuboot/boot/zephyr
-- Zephyr version: 3.0.0-rc1 (~/work/zephyrproject/zephyr), build: v3.0.0-rc1-4-g3a979bb81904
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Found west (found suitable version "0.11.0", minimum required is "0.7.1")
-- Board: efr32_radio_brd4104a
-- Cache files will be written to: ~/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.13.1 (~/.local/opt/zephyr-sdk-0.13.1)
-- Found dtc: ~/.local/opt/zephyr-sdk-0.13.1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.13.1 (~/.local/opt/zephyr-sdk-0.13.1)
-- Found BOARD.dts: ~/work/zephyrproject/zephyr/boards/arm/efr32_radio/efr32_radio_brd4104a.dts
-- Found devicetree overlay: ~/work/zephyrproject/mcuboot/boot/zephyr/dts.overlay
-- Generated zephyr.dts: ~/work/zephyrproject/mcuboot/boot/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: ~/work/zephyrproject/mcuboot/boot/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: ~/work/zephyrproject/mcuboot/boot/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: ~/work/zephyrproject/mcuboot/boot/zephyr/build/zephyr/dts.cmake
Parsing ~/work/zephyrproject/mcuboot/boot/zephyr/Kconfig
Loaded configuration '~/work/zephyrproject/zephyr/boards/arm/efr32_radio/efr32_radio_brd4104a_defconfig'
Merged configuration '~/work/zephyrproject/mcuboot/boot/zephyr/prj.conf'
Configuration saved to '~/work/zephyrproject/mcuboot/boot/zephyr/build/zephyr/.config'
Kconfig header saved to '~/work/zephyrproject/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: ~/.local/opt/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
MCUBoot bootloader key file: ~/work/zephyrproject/mcuboot/root-rsa-2048.pem
CMake Error at CMakeLists.txt:312 (string):
  string sub-command FIND requires 3 or 4 parameters.
```

The issue was introduced by #1255.

mnkp, 2022-01-27:
@wcappelle

wcappelle, 2022-01-28:
@mknp, thanks for pinpointing. Will look into this

wcappelle, 2022-01-28:
@mnkp Can you check the PR if it solves your problem? I've tested this on my branch, and should fix the problem of cmake.
Build didn't succeed on this board config due to overflow of the partition size. I don't think this is a problem I introduced. Let me know if I can be of any further assistance.

utzig, 2022-01-28:
I reproduced the problem yesterday and the PR solved it.

mnkp, 2022-01-28:
> I've tested this on my branch, and should fix the problem of cmake.
Build didn't succeed on this board config due to overflow of the partition size.

Thanks for fixing the issue. Sorry, I should have used a different board as an example. I forgot that with the newer Zephyr versions the MCUboot doesn't fit into bootloader partition on this specific board anymore.

PR #1298 (2022-01-28): zephyr: fix compilation when no CONFIG_BOOT_ENCRYPTION_KEY_FILE
Fix compilation of zephyr builds when there is no CONFIG_BOOT_ENCRYPTION_KEY_FILE set.

Fixes #1297
closed 2022-01-28 by utzig