mcu-tools/mcuboot #1300: Correct syntax for imgtool.py command

PxGreen101, 2022-02-03:
I am using imgtool to create a signed image for Zephyr. I am using the zephyr.bin file as my image input file and the command syntax is as follows:

./imgtool.py sign --key ../root-rsa-2048.pem --header-size 0x200 --align 8 --version 1.3 --slot-size 0x32000 zephyr.bin zephyrn1out.bin

The zephyr1out.bin file is generated. Can someone confirm this is the correct format to sign a file?

The size of my image slot is 0x32000

I am using mcumgr to update the firmware but the image does not update even though the swap and copy seems correct.

utzig, 2022-02-03:
Yes, it's correct.

utzig, 2022-02-03:
The key must be the same built in the bootloader, the `--header-size` must match but this value is probably correct, you should use `west sign` instead of fiddling with `imgtool` directly.

PxGreen101, 2022-02-03:
Ok I thought it may be correct. Do I need to add --pad or --confirm ? I noticed these increase the size of the output file significantly. 

I am using the STM32G0 Nucleo board and it already has the MCU boot and the primary application running on it. I can successfully update the new application using mcumgr into the second slot.

I can confirm the image as well, however the image does not update

[image]

I have also opened up a Zephyr issue regarding this:
https://github.com/zephyrproject-rtos/zephyr/issues/42453

utzig, 2022-02-03:
> Ok I thought it may be correct. Do I need to add --pad or --confirm ? I noticed these increase the size of the output file significantly.

`--pad` and `--confirm` are required only if you aren't using `mcumgr`, but since you are, then what you do is `mcumgr image confirm <sha>`, or `mcumgr image test <sha>`.

Your second screen makes it look like it run a successful upgrade. Are you writing the exact same image on both slots? Maybe with a different version to generate a different hash?

Try running `mcumgr image test` and check it does a `Swap type: test` and after a reset `Swap type: revert`.

PxGreen101, 2022-02-03:
I have created 2 different images. The first image prints Hello World version1 and the second image has been modified to print Hello World version2. They are 2 different images  

I have tried with mcumgr image test previously but that did not work either. I tried it again in order to show the output

[image]

[image]

utzig, 2022-02-03:
What does it log when you reboot again after the steps above?

PxGreen101, 2022-02-04:
Same thing. I can reboot any number of times
[image]

utzig, 2022-02-04:
How are you building the bootloader?

PxGreen101, 2022-02-04:
From the mcuboot/boot/zephyr directory, build using
/bootloader/mcuboot/boot/zephyr$ west build -b nucleo_g0b1re -p

PxGreen101, 2022-02-04:
As mentioned I have opened a Zephyr issue regarding this. I wanted to be sure that I was creating the signed images correctly which appears to be the case. Thanks for confirming utzig.

PxGreen101, 2022-02-04:
By the way I did the exact same steps and targeted the Nucleo STM32F429ZI board instead. Same MCUboot bootloader, same application, same configuration options, same commands when building and signing with imgtool. Everything works correctly with the Nucleo STM32F429ZI. There must be some issue to target the Nucleo STM32G0B1RE.

utzig, 2022-02-04:
> There must be some issue to target the Nucleo STM32G0B1RE.

I was thinking the same when reading your Zephyr issue. If I had a board I would take a look, but probably someone at ST will. Or you might do some debugging of the G0 flash driver yourself! :-)

PxGreen101, 2022-02-08:
I believe this is a Zephyr driver issue with the STM32G0B1RE driver so I am closing this particular ticket.

closed 2022-02-08 by PxGreen101