mcu-tools/mcuboot #1306: Integration with TPM 2.0?

tomoveu, 2022-02-15:
Congratulations on fork number 400  as I just found out about the project and forked it to try out in my free time.

Is there a plan to integrate with hardware security modules(HSM), especially a TPM?

Kudos on this valuable Open-source project 💯

d3zd3z, 2022-02-28:
I guess the answer depends on what you mean by integration with HSM. Issue #599 covers adding HSM support to the signing tool `imgtool`. This will be useful for production as it would be better to not have to have the signing key just sitting on a PC.

If instead, you are referring to using TPM hardware on the target, that currently isn't in any plans or roadmap. Are you think of having this used for measurements during boot. There is some support for this with TF-M, but I'm not aware of any work doing this with hardware-based security.

tomoveu, 2022-03-01:
@d3zd3z the latter, TPM on the target. Doing measurements during boot and establishing a root of trust for reporting (and even storage).

If I understand correctly no one has asked for such functionality and it's not in the current plans.

What do you mean by your last sentence that you don't know anyone doing work with TF-M and hardware-based security?

Could you please elaborate.

d3zd3z, 2022-03-03:
> What do you mean by your last sentence that you don't know anyone doing work with TF-M and hardware-based security?

I think just what you are saying, nobody has asked for such functionality and it is not in the current plans.

tomoveu, 2022-03-04:
Out of curiosity, what would be the path if someone wants to contribute it to the project? 

Is there some planning/alignment involved?

What is a good target/board to do such mcuboot work?

d3zd3z, 2022-03-07:
It is probably easiest to join our slack (mentioned on mcuboot.com). You can also join the TSC call which happens 2 Tuesdays a month. The mailing list at groups.io has the calendar with the invite. You do have to sign up (even if you elect not to receive email), otherwise, the meeting ID gets out and we get zoom bombed.

Target probably depends on the target you have, since I don't think any of the common targets we support have TPMs on them.

tomoveu, 2022-09-05:
@d3zd3z thank you for the detailed information. I would try to join a call in the next month. Feel free to close this as resolved and perhaps extend the readme with a short note about the status of TPM support.

By the way, we have our [TPMdev](https://developers.tpm.dev/posts/tpmdev-2022-conference) developers conference on 13th of October. Please join in. Recordings from previous years you could find on Vimeo, [for example](https://vimeo.com/644241251).

Once again thank you for the good info and sorry for the delay in my response.

tomoveu, 2023-02-24:
Closing this issue because it was already discussed at length.

closed 2023-02-24 by tomoveu