mcu-tools/mcuboot #1308: MCUboot boot-up time

aniruddhanalgirkar-eaton, 2022-02-18:
Hi,
Recently we deployed the MCUboot in one of our device. In this, we selected the SWAP method, BOOT_VALIDATE_SLOT0 is selected as n.
so, under this configuration, I called the API, k_uptime_get(); just before the application is getting called in the MCUboot main.c file. And thus, I measure the boot-up time as 31 ms. I want this to be lesser than this. (assume, 15 - 21 ms could be acceptable).
So, how much minimum can it be ? How to optimize it further ?
Thanks for any additional suggestions !

nvlsianpu, 2022-02-18:
Can you turn off LOG and measure the timing using a gpio? I suspect that most of the time you had measured is just needed for log transmission,.

d3zd3z, 2022-02-25:
Also, what kind of CPU does your device have? A Cortex-M0 will be a lot slower than a larger device would be.

Unless you have a particularly unusual flash device, I would recommend SWAP_MOVE instead of SWAP_SCRATCH. There is no swap partition needed, but the slot 0 should be 1 sector larger than slot 1.

aniruddhanalgirkar-eaton, 2022-03-04:
the SWAP_MOVE option greatly reduced the boot-up time. 
Next, I am going to try with "config BOOT_TIME_MEASUREMENT" to measure it better way. Thanks !

closed 2022-09-15 by github-actions[bot]