mcu-tools/mcuboot #1309: Drag and Drop programming support

beriberikix, 2022-02-18:
Has anyone explored drag and drop programming? This mode of programming very popular for the platforms targeting education and hobbyists, but is also appealing for rapid prototyping workflows. Some silicon now even has support in ROM (like the RP2040.) 

The way it is achieved is by leveraging a Mass Storage device. The user builds the application, drags it to a drive and the bootloader automatically reboots the device when detected. 

See reference:
- https://github.com/microsoft/uf2
- https://learn.adafruit.com/adafruit-feather-m0-express-designed-for-circuit-python-circuitpython/uf2-bootloader-details
- https://github.com/mmoskal/uf2-stm32f
- https://github.com/adafruit/tinyuf2

Is DnD programming feasible with MCUboot? Has anyone explored it as an option?

d3zd3z, 2022-02-25:
I think it is as reasonable of an option as the current USB DFU. But, I don't know of an implementation that could just be dropped in.

beriberikix, 2022-02-26:
I don't know of any drop-in solutions either. But probably the mostly viable USB stack is probably [TinyUSB](https://github.com/hathach/tinyusb). See hathach/tinyusb#1355. There's a [UF2 bootloader](https://github.com/adafruit/tinyuf2) that could be used as a reference.

Thoughts?

closed 2022-12-05 by github-actions[bot]

josuah, 2024-02-28:
Sorry for recovering this old issue, but for anyone curious, it is possible to use TinyUF2 as a library.
I never had the time to upstream the few changes needed though.
<https://github.com/adafruit/tinyuf2/discussions/272>

It goes through using TinyUF2 as a submodule (for instance) and only compiling selected files, providing others from the outside such as `main.c` (to remove main() out of it) for instance.

tannewt, 2026-03-04:
FYI, I'm working on it.

josuah, 2026-03-04:
I really like the idea but am not sure MCUboot maintainers (not me) would want to integrate and maintain an USB stack integration as part of MCUboot (that was probably not your goal).

This seemed like going beyond the [basic recovery protocol](https://docs.mcuboot.com/serial_recovery.html) integrated, while advanced upload protocols exist (USB, Bluetooth, CAN, Serial...) in an application payload (part of the main application or dedicated image).

Do you think such an approach would work? This seems low-effort on both side: just plugging to the existing APIs of MCUboot to provide a "device firmware upgrade" payload (a "upgrade" mode selected when a chosen "BOOT" GPIO is active), which would fill a new image in a MCUboot partition slot ready to be picked by MCUboot.

I just thought of this in the back of my head and meant to share it in hope this was useful, possibly something you already knew...