mcu-tools/mcuboot #1312: zephyr/single_loader #if 0 in image decryption code.

nvlsianpu, 2022-02-21:
https://github.com/mcu-tools/mcuboot/blob/d8eff810ad2ce08c66764560a82f415ca710e3f9/boot/zephyr/single_loader.c#L354

What we can do about that.

nvlsianpu, 2022-02-21:
cc @@wcappelle

wcappelle, 2022-02-21:
This step is skipped since it was introducing some extra latency.
Skipping this means that the image is not first validated before decryption is started. So decryption is done, and after decryption is finished, I rely on the normal boot process to validate the decrypted image. I don't think this is a bad thing since an attacker still needs to have the private key to sign the image if he wants to run this.
If you think this is a security risk, I would suggest to always enable it. During my development I had it initially enabled, but due to above reasoning, i've disabled it since i was working on a slow cortex-M0.

nvlsianpu, 2022-02-21:
@wcappelle I mean that this is no configurable, but just hardcoded. We should have a configuration option instead of `#if 0`

nvlsianpu, 2022-02-23:
This excluded code gives possibility for reject either badly signed and badly encrypted image before non volatile decryption will happens. But gain is negligible this is USB DFU - someone have already the way for writing to the image slot anytime that someone want.

For reference: Without the encryption, the image validation might be performed only before application call

Also a user might disabled image validation - if we make this code arbitrary ON this will introduce unwanted procedure. For good security level.

I would remove that dead code.

wcappelle, 2022-02-23:
I'm also in favor of removing. I only left the code there since I was still in doubt what reviewers would want wrt to security concerns. But I follow the same reasoning.

closed 2022-09-06 by github-actions[bot]