mcu-tools/mcuboot #1342: Encrypting/decrypting firmware image

Nano-HK, 2022-04-08:
Hi all, 

I have been trying to make decrypting of firmware image on MCUboot but have not been successful. 
**Platform**: Zephyr
**Board**: nrf5340dk_nrf5340_cpuapp

This is what I've done;

**On the bootloader**
- Built the MCUboot bootloader with signature key and encryption key, `CONFIG_BOOT_ENCRYPT_EC256=y` `CONFIG_BOOT_SIGNATURE_TYPE_ECDSA_P256=y` using the default keys `root-ec-p256.pem` for signature and `enc-ec256-priv.pem` for encryption.

**On the application**
- Built one application with only signature and no encryption of firmware image --> this works fine!
- Built the other application with the same signature key mentioned above and the public key of `enc-ec256-priv.pem` --> **this does not work**

On the device, I try to do upgrade by swaping secondary image slot to the primary image slot - to do this I'm using Zephyr shell commands to do boot_upgrade swap type: permanent. In debugging, the mcuboot goes to `boot_enc_load` to load the encryption key.

What am I missing? Why can't the mcuboot decrypt the firmware image? 
How can I decrypt the firmware image, what is the process? 

Would appreciate any tips and help on how to proceed with this. 

Regards,
Nano

nvlsianpu, 2022-04-08:
> ... and the public key of enc-ec256-priv.pem

What you mentioned is the private key. You should use the public key with imgtool.

> -E, --encrypt filename          Encrypt image using the provided public key.
                                  (Not supported in direct-xip or ram-load
                                  mode.)

Nano-HK, 2022-04-11:
Sorry, for clarification I am using the public key, enc-ec256-pub.pem. But it is still not working. 

Do you have proposal on how to setup a decrypt firmware image test - what mode and configurations to have? 

Regards,
Nano

closed 2022-10-23 by github-actions[bot]