mcu-tools/mcuboot #1347: boot_copy_region() trying to read past secondary slot

rmsaad, 2022-04-12:
I am using the swap move algorithm to swap images. In the documentation it states :
> Therefore the most memory-size-effective slot layout is when the primary slot is exactly one sector larger than the secondary slot

Therefore my primary image is exactly 1 sector larger than the secondary image.

> .fa_size = (260 * 1024),                    // (primary img)

vs.

> .fa_size = (256 * 1024),                   // (secondary img)

My problem is that the boot_copy_region() function in loader.c will try to read from the secondary slot using the primary slot's image size and then fail trying to read an out of bounds area. To verify, if I change the size of the primary slot to match the secondary slot it will function correctly. 

Any ideas would be appreciated.

rmsaad, 2022-04-14:
I've done some more exploring and this problem seems to only occur when the image is copied from the secondary slot to primary slot. There is no problem with swapping 2 discreet images between the primary and secondary slot.

Could I have misinterpreted the documentation in configuring flash_area(s) for swap move algorithm? Is the 1 sector padding at the end of the primary slot achieved some other way?

d3zd3z, 2022-05-05:
The primary slot should be one erase sector larger than the secondary slot, however, the maximum image still needs to fit in the space available in the second slot. The first step of swap move is to "slide" the image down by one slot. If you are finding it going beyond the bounds of a slot, this could be a bug in the code that detects an image being too large.

rmsaad, 2022-05-05:
[image]

I stumbled across this problem while trying to verify that my encrypted image were decrypting correctly. So I erased the image in the first slot and programed the encrypted image in the second slot and reset my board. Since mcuboot was copying instead of swapping an assertion fails for the boot_copy_image() function.

There is no problem with swapping however and also no problem copying if both slots are the same size.

closed 2022-05-05 by rmsaad