mcu-tools/mcuboot #1348: ECIES not working with AES 256 keys

michaelthomasj, 2022-04-13:
The signing script seems to hardcode the key length to 128 for ECIES: 
https://github.com/mcu-tools/mcuboot/blob/93d02179b32056320deaac7919247173793febbe/scripts/imgtool/image.py#L287
The HKDF length is 48 there: 16 for the AES key and 32 for the MAC. That is correct for AES 128. 
The HKDF should be 64; 32 each for AES and MAC. The line after that should also be updated to use the first 32 bytes for the AES key.

d3zd3z, 2022-05-05:
I'm not sure the rest of the code has been tested with anything other than AES-128. It'd be worth fixing if we add support for a larger cipher, maybe even earlier, since it would not be obvious this needs to be fixed.

michaelthomasj, 2022-05-05:
I agree that since we only support ECIES-P256 and RSA-OAEP 2K/3K, AES 128 is sufficient. However, if MCUboot plans to support EC-P384 then we would have to support AES 256 or 192. 
Since we do not support AES 256, we should remove the "--encrypt-keylen" option from imgtool since it lets the user select AES 128 or 256 but then it just implements AES 128 which is misleading.

closed 2022-11-17 by github-actions[bot]